dl9rdz/rdz_ttgo_sonde #1: Config for new master version

DL2MF, 2019-04-15:
Hi Hans-Peter,

really a great and very nice project, congrats! 

I adapted your latest dev branch to my TTGO ESP32 LoRa V2.1 board (uses port 21/22 for the OLED). Maybe we can make some config additions for several boards in your new config.
Can you please upload the latest config to the data directory.

73 Meinhard "Mike", DL2MF

DL2MF, 2019-04-16:
OK, found in todays devel-branch under construction ;-)

dl9rdz, 2019-04-16:
Started some code reorg today to make this configuration configurable. Not yet finished, and I will be away on holidays for the next days, so progress will take some time... But I also ordered a v2.1 board from China two weeks back (which should arrive here next week), so I will also be able to test the code with that board.

sp3rat, 2019-04-17:
Hi Hans-Peter,
I am sorry my English.
I like your project rdz_ttgo_sonde.
I have one question. How to setup my IP as static?
Because  the ttgo many times do not working with my wifi network.
73! Jurek, de sp3rat

dl9rdz, 2019-04-18:
Currently in Station Mode IP is always dynamic. Change probable possible in source code, will put this on my todo list
In AP mode IP is always 192.168.4.1

Network is always fine with my Android AP but needs several tries with my Fritzbox AP. Will have a look how to improve this when back in Passau next week

dl9rdz, 2019-04-29:
WiFi networking should now be stable (I hope), closing this issue.
If there is a real need for static IP addresses in station mode let me know.

closed 2019-04-29 by dl9rdz