dl9rdz/rdz_ttgo_sonde #3: Wifi problem

b4sh, 2019-04-26:
Hello, first thanks for nice Open Source TTGO project.

If we have more than one wifi network in range after scan loop pw is always change to "test" and wifi not connect.

This is my patch for this issue.

https://github.com/b4sh/rdz_ttgo_sonde-1/commit/01e5ce74e625b2ced5600823aa8780716f94e81e#diff-3f28f753b8cab0c9c3b5a092e540dc12

dl9rdz, 2019-04-28:
Should be fixed in v0.5

closed 2019-04-28 by dl9rdz

b4sh, 2019-04-28:
Now, new background mode not work for me, endless scanning. Switching wifi mode to "3" also not work. I have 16 wifi networks in range (urban area), maybe this matter. Code after if (index >= 0) { // some network was found (.......) not exec - index is always "-1".  curidx also is always "-1".