dl9rdz/rdz_ttgo_sonde #7: WiFi v0.5

b4sh, 2019-04-28:
Now, new background mode not work for me, endless scanning. Switching wifi mode to "3" also not work. I have 16 wifi networks in range (urban area), maybe this matter. Code after if (index >= 0) { // some network was found (.......) not exec - index is always "-1". curidx also is always "-1".

dl9rdz, 2019-04-28:
Hmm. I unfortunately cannot reproduce this (I have less then 16 wifi networks here, but this shouldn't be a problem...).

Can you copy the console output from Arduino IDE (starting with the line "x networks in network.txt" and until "Running main loop.")?

b4sh, 2019-04-28:
Wifi mode 1

`2 networks in networks.txt

RDZsonde: RDZsonde
WIFI-LINK: secretpw <<< ------- My Network

RX LNA Gain is -999
SX1278 configuration finished
Setup finished

Reading channel config:
Add 402.299988 - type 2 (on/off: 1)- Site: 
Add 402.500000 - type 2 (on/off: 0)- Site: 
Add 402.700012 - type 2 (on/off: 1)- Site: 
Add 403.000000 - type 2 (on/off: 0)- Site: 
Add 404.100006 - type 2 (on/off: 1)- Site: 
Add 404.299988 - type 2 (on/off: 0)- Site: 
Add 404.500000 - type 2 (on/off: 0)- Site: 
Add 404.700012 - type 2 (on/off: 0)- Site: 
Add 405.100006 - type 2 (on/off: 0)- Site: 
Add 405.700012 - type 2 (on/off: 1)- Site: 
Add 405.899994 - type 2 (on/off: 1)- Site: 
Add 405.100006 - type 2 (on/off: 1)- Site: 
Add 405.299988 - type 2 (on/off: 0)- Site: 
Add 403.329987 - type 1 (on/off: 0)- Site: 
Add 403.450012 - type 1 (on/off: 0)- Site: 
Add 403.470001 - type 1 (on/off: 0)- Site: 
Add 403.850006 - type 1 (on/off: 0)- Site: 
Add 403.869995 - type 1 (on/off: 0)- Site: 
Add 403.890015 - type 1 (on/off: 0)- Site: 
Setting up receiver on channel 0
Setup sx1278 for RS41 sonde
Exact bitrate is 4799.98
Setting SX1278 config for RS41 finished

RS41: setting RX frequency to 402300000.00
Running main loop. free heap:285612
  RSSI: 0 LNA Gain: 0
WifiBackground: state 0
WLAN start scan
[WiFi-event] event: 0
WiFi interface ready
Running main loop. free heap:248272
[WiFi-event] event: 2
WiFi client started
TIMEOUT
Scanner: receiveFrame returned: 1
Setting up receiver on channel 2
Setup sx1278 for RS41 sonde
Exact bitrate is 4799.98
Setting SX1278 config for RS41 finished

RS41: setting RX frequency to 402700000.00
  RSSI: 0 LNA Gain: 0
WifiBackground: state 1
Running main loop. free heap:247616
TIMEOUT
Scanner: receiveFrame returned: 1
Setting up receiver on channel 4
Setup sx1278 for RS41 sonde
Exact bitrate is 4799.98
Setting SX1278 config for RS41 finished

RS41: setting RX frequency to 404100000.00
  RSSI: 0 LNA Gain: 0
WifiBackground: state 1
Running main loop. free heap:246944
[WiFi-event] event: 1
Completed scan for access points
[... 188 lines cut ...]
[WiFi-event] event: 3
WiFi clients stopped
Activating access point mode
[WiFi-event] event: 0
WiFi interface ready
[WiFi-event] event: 3
WiFi clients stopped
[WiFi-event] event: 13
WiFi access point started
[WiFi-event] event: 14
WiFi access point  stopped
[WiFi-event] event: 13
WiFi access point started
[WiFi-event] event: 13
WiFi access point started
AP IP address: 192.168.4.1
  RSSI: 0 LNA Gain: 0
WifiBackground: state 4
Running main loop. free heap:232436
TIMEOUT
Scanner: receiveFrame returned: 1
Setting up receiver on channel 2
Setup sx1278 for RS41 sonde
Exact bitrate is 4799.98
Setting SX1278 config for RS41 finished`

dl9rdz, 2019-04-28:
Very strange. 

In most recent version in the "devel" branch I have added a bit more debug output. Maybe you can retry with that (only with wifi=3, the background mode with wifi=1 has little debug output)

The expected output is something like:
23:34:27.046 -> 2 networks in networks.txt
23:34:27.046 -> 
23:34:27.046 -> RDZsonde: RDZsonde
23:34:27.046 -> DinoGast: Schokolade
[skipping some lines]
23:34:29.928 -> Completed scan for access points
23:34:29.928 -> Network name: Dinosauro
23:34:29.965 -> Signal strength: -56
23:34:29.965 -> MAC address: E0:28:6D:4E:FE:AC
23:34:29.965 -> Encryption type: WPA_WPA2_PSK
23:34:29.965 -> -----------------------
23:34:29.965 -> No match: 'Dinosauro' vs 'RDZsonde'
23:34:29.965 -> No match: 'Dinosauro' vs 'DinoGast'
23:34:29.965 -> Network name: DinoGast
23:34:29.965 -> Signal strength: -56
23:34:29.965 -> MAC address: E2:28:6D:4E:FE:AC
23:34:29.965 -> Encryption type: WPA2_PSK
23:34:29.965 -> -----------------------
23:34:29.965 -> No match: 'DinoGast' vs 'RDZsonde'
23:34:29.965 -> Match found at scan entry 1, config network 1
23:34:29.965 -> Network name: LEDnetB94524
23:34:30.002 -> Signal strength: -79
23:34:30.002 -> MAC address: 62:01:94:B9:45:24
23:34:30.002 -> Encryption type: Open
23:34:30.002 -> -----------------------
23:34:30.002 -> No match: 'LEDnetB94524' vs 'RDZsonde'
23:34:30.002 -> No match: 'LEDnetB94524' vs 'DinoGast'
23:34:30.002 -> Connecting to: DinoGast

So for every network found by scan, you should get some lines with "No match..." or "Match found..", and there should be one match, followed by "Connected to:" the matching network

b4sh, 2019-04-28:
```
Network name: WIFI-LINK
Signal strength: -34
MAC address: 78:24:AF:E4:C1:90
Encryption type: WPA2_PSK
-----------------------
No match: 'WIFI-LINK' vs 'RDZsonde'
No match: 'WIFI-LINK' vs 'WIFI-LINK'
```

dl9rdz, 2019-04-28:
Well, thanks... but still, I am somewhat clueless.
This part of the code is quite simple:

```
if (strcmp(id, networks[i].id.c_str()) == 0) {
      Serial.printf("Match for %s at %d\n", id, i);
      return i;
}
Serial.printf("No match: '%s' vs '%s'\n", id, networks[i].id.c_str());
```

Now in your output, both strings **look** identical, and strcmp should return 0 in that case, and **not** print "No match".

Maybe one of your hyphens (in your SSID? in network.txt?) is not an iso8859-1 hyphen but some other utf-8 hyphen? And gets magically converted to an iso8859-1 hyphen when copy&pasting the text to this web page? Or one of the two contains some other non-printable character? 

Most recent devel contains even more debug messages... maybe this helps.  Otherwise I have no more ideas.

b4sh, 2019-04-28:
I found the reason - line endings. My file somehow changed to Windows CR LF, after change to Unix LF everything works. 

Thanks for your assist. It is a pity that I did not get it earlier. This is worth mentioning in the readme file. It may save some time to others. :)

Now i have ANSI encoding with Unix LF line endings.

dl9rdz, 2019-04-29:
Also found some minor C++ errors that could possibly cause network problems. Fixed in the latest release.

closed 2019-04-29 by dl9rdz

ok2it, 2019-10-30:
I confirm All Works on 64 plus Network Overhere

dl9rdz, 2019-11-03:
The latest devel update makes all data files tolerant to line endings, so CR LF should be fine now instead of just LF.