dl9rdz/rdz_ttgo_sonde #9: M10 sonde [enhancement]

b4sh, 2019-06-11:
Is it possible to add M10 sonde to decoder? If needed i can send working M10 sonde to author.

dl9rdz, 2019-06-13:
I don't see any M10 in my location (the closest launch site with M10, Vienna, is >300km away), but if I get a working M10 and have some spare time available, I can give it a try. No promise :-) My address: Hansi Reiser, DL9RDZ, Stephanstr. 151, 94034 Passau, Germany.

It should be possible to receive M10 with the TTGO LORA, as F5ZJ seems to have a working implementation. Unfortunately I am unable to contact him (contact form on his site http://keraman.free.fr/ is dysfunctional, and I did not find any other way to contact him).

sp3rat, 2019-06-15:
Hello  :)
It is e-mail to F5ZJ  kera5@free.fr
Jurek

b4sh, 2019-06-17:
It turned out that my M10 sonde is damaged. When I hunt a new one, I will prepare a package.

oe3jtb, 2019-07-08:
Hi Hansi, still open your M10 request? If yes , you have friends at Schärding or other near OE town? 73 Alex, OE3JTB

oe6pwe, 2019-07-09:
Hello,
Contact me I can send you a M10.
73 Wolfgang, OE6PWE

ok2it, 2019-07-11:
mee too , let me know if needed.

dl9rdz, 2019-07-21:
Thanks to Wolfgang for sending me two M10 👍 

I will see what I can do.... might take a few weeks though as I am about to leave for holidays beginning of August, and then in September I am organizing the next HAMNET meeting in Passau (and a few other things...)

dl9rdz, 2019-11-01:
I have added some code for M10 in devel20191101. Not much tested, but it decodes the two M10s I got :-)
Unlike the french software, it does not require wiring SX127x's DIO2 to an ESP32 GPIO pin, but instead attempts to use the FIFO, even if the modulation/frame format is not a good match to the RF chip's features... Not sure how reliable this works with weak signals though.
IDs are dxl new style.

oe3jtb, 2019-11-02:
Hansi tnx for implementing M10, OE8JPQ and myself did a first test very fine thank you!!!

dl9rdz, 2019-11-10:
Seems to work fine, so closing this issue :-)

closed 2019-11-10 by dl9rdz