dl9rdz/rdz_ttgo_sonde #10: New function - APRSdroid

Piter-NS, 2019-08-22:
Is it possible to add function which can send data received from radiosonde trought bluetooth to APRSdroid app. I need this to track the position of the radiosonde live while I drive my car to it's landing point.

dl9rdz, 2019-09-11:
I tried to implement this with internal bluetooth, but the standard bluetooth library (BluetoothSerial) consumes about 500k memory, which is more than available in the current configuration.

If anyone knows a more lightweigth bluetooth library, I might consider this in the future.

I might also implement a bluetooth interface with external hardware (internal bluetooth would most likely severly interfere with internal WiFi, so probably its the better option anyway). Maybe if more spare time is available.

What I added in the latest devel version (not much tested yet) is a KISS TNC over TCP/IP interface, which can be used in combination with APRSdroid.

You need to activate "KISS TNC" in the TTGO configuration, and on APRSdroid you can select APRS connection => Protocol: TNC (KISS), Type of connection TCP/IP, Server <IP address of the TTGO), and you will receive a position update after each decoded frame (which probably is fine for RS41, but a bit too much for DFM as not each frame contains a new position information...)

Data contains only ID, lon/lat and alt, everything else is just for testing right now (will change some time in future). Please use this interface only for connecting APRSdroid to TTGO, not for feeding data to the internet! This would require somewhat more polished data...

sp3rat, 2019-09-29:
Tnc tcp/ip for Aprsdroid it is Ok  :)
It is working very fine

Piter-NS, 2019-11-29:
How to activate KISS TNC? 
What to enter in the KISS TNC field because I entered 1 or the port number (14590) and after restarting TTGO I always have the value 0 in this field.

Firmware version: devel20191128

yeckel, 2019-11-29:
I'm setting it to 1 in `{"kisstnc", "KISS TNC (port 14590) (needs reboot)", 1, &sonde.config.kisstnc.active},` then it works. It's just a workaround.

Piter-NS, 2019-11-29:
After entering 1 in the KISS TNC (port 14590) (needs reboot) field and pressing UPDATE, I reboot TTGO and after connecting the KISS TNC field is again 0

dl9rdz, 2019-11-29:
Yes you found a bug that I don't know got into the current version. Will fix it later today.

closed 2020-09-02 by Piter-NS