dl9rdz/rdz_ttgo_sonde #13: Turn on GPS on new T-BEAM

ok1cdj, 2019-09-30:
Hi, there is new version of T-Beam which have power management chip. You need turn on GPS on. Here is example code which working for me outside the project. When i am trying to implement it to code, there is problem when i am use Wire.begin(21, 22);, the I2C display stop working, but GPS works. When i am not using Wire.begin(21, 22); the comunication with power management chip fails, but display is working. There is some problem with display and power management chip on same i2c bus, any idea how to solve it ? 

73 Ondra OK1CDJ
```
#include <axp20x.h>

AXP20X_Class axp;

void setup()
{
  Serial.begin(115200);
  Wire.begin(21, 22);
  if (!axp.begin(Wire, AXP192_SLAVE_ADDRESS)) {
    Serial.println("AXP192 Begin PASS");
  } else {
    Serial.println("AXP192 Begin FAIL");
  }
  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON);
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
}
```

dl9rdz, 2019-09-30:
I guess the problem is that the Display code uses software I2C and your initialisation code hardware I2C, and doing both with the same ports is likely to cause problem.

Replacing U8X8_SSD1306_128X64_NONAME_SW_I2C with U8X8_SSD1306_128X64_NONAME_HW_I2C might solve the problem (not tested, I don't have a v1.0 board yet)

ok1cdj, 2019-10-01:
Tested with U8X8_SSD1306_128X64_NONAME_HW_I2C , but display not working.. need more time to investigate it..

dl9rdz, 2019-10-05:
I have a working version now, will commit it after some more testing
- you need to set voltage output to 3.3V
- U8X8_SSD1306_128X64_NONAME_HW_I2C has parameters in a different order than SW_I2C

dl9rdz, 2019-10-05:
T-Beam 1.0 initialization should be working in most recent devel version

closed 2019-10-05 by dl9rdz