dl9rdz/rdz_ttgo_sonde #14: LoRa ESP32 v2 SX1278 is not working since v0.7.1

istamov, 2019-10-14:
Hello,

First of all thank you very much for your great project, it is really useful and really fun to play with :)

I have an issue with the following module - LoRa ESP32 v2 0.96" Blue OLED Display SX1278 (https://www.aliexpress.com/item/32829056144.html).
Since version 0.7.1 and all of the development versions afterwards, the display on it is not working, it is just blank after flashing.

Here is the log from the serial port:
```
0:1 1:0 2:0 3:1 4:0 5:1 6:0 7:1 8:0 9:1 10:1 11:1 12:0 13:0 14:1 15:0 16:0 17:0 18:0 19:0 20:0 21:1 22:0 23:0 24:0 25:0 26:0 27:0 28:0 29:0 30:0 31:0 32:0 33:0 34:0 35:0 36:0 37:0 38:0 
0:1 1:1 2:0 3:1 4:0 5:1 6:0 7:1 8:0 9:1 10:1 11:1 12:0 13:0 14:1 15:0 16:0 17:0 18:0 19:0 20:0 21:1 22:0 23:0 24:0 25:0 26:0 27:0 28:0 29:0 30:0 31:0 32:0 33:0 34:0 35:0 36:0 37:0 38:0  (before setup)
Autoconfig: looks like v1 board
Initializing SPIFFS
Reading initial configuration
configuration option 'led_pout'=9 
configuration option 'gps_txd'=-1 
configuration option 'maxsonde'=20 
configuration option 'debug'=0 
configuration option 'wifi'=3 
configuration option 'kisstnc.active'= 1 
Invalid config option 'kisstnc.a 

!
I2C device found at address 0x02 !
I2C device found at address 0x03 !
I2C device found at address 0x04 !
I2C device found at address 0x05 !
I2C device found at address 0x06 !
I2C device found at address 0x07 !
Unknow error at address 0x08
I2C device found at address 0x09 !
I2C device found at address 0x0A !
I2C device found at address 0x0B !
Unknow error at address 0x0C
I2C device found at address 0x0D !
I2C device found at address 0x0E !
I2C device found at address 0x0F !
I2C device found at address 0x10 !
I2C device found at address 0x11 !
I2C device found at address 0x12 !
I2C device found at address 0x13 !
I2C device found at address 0x14 !
I2C device found at address 0x15 !
I2C device found at address 0x16 !
I2C device found at address 0x17 !
I2C device found at address 0x18 !
I2C device found at address 0x19 !
I2C device found at address 0x1A !
I2C device found at address 0x1B !
I2C device found at address 0x1C !
I2C device found at address 0x1D !
I2C device found at address 0x1E !
I2C device found at address 0x1F !
I2C device found at address 0x20 !
I2C device found at address 0x21 !
I2C device found at address 0x22 !
I2C device found at address 0x23 !
I2C device found at address 0x24 !
I2C device found at address 0x25 !
I2C device found at address 0x26 !
I2C device found at address 0x27 !
I2C device found at address 0x28 !
I2C device found at address 0x29 !
I2C device found at address 0x2A !
I2C device found at address 0x2B !
I2C device found at address 0x2C !
I2C device found at address 0x2D !
I2C device found at address 0x2E !
I2C device found at address 0x2F !
I2C device found at address 0x30 !
[... 221 lines cut ...]
Showing welcome display
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
tou1
y
2 networks in networks.txt

RDZsonde: RDZsonde
DinoGast: Schokolade
before disp.initFromFile... layouts is 0x3ffbf20cReading from /screens.txt. available=5439
Line: '# Definition of display content and action behaviour'
Line: '#'
Line: '# Timer: (view timer, rx timer, norx timer)'
Line: '# - value -1: timer '

SX1278 configuration finished
Setup finished
```

dl9rdz, 2019-10-14:
please check devel20191014b

istamov, 2019-10-15:
Thank you, but I am getting the same result, blank screen.

```
Tim?0:1 1:0 2:0 3:1 4:0 5:1 6:0 7:1 8:0 9:1 10:1 11:1 12:0 13:0 14:1 15:0 16:0 17:0 18:0 19:0 20:0 21:1 22:0 23: 
0:1 1:1 2:0 3:1 4:0 5:1 6:0 7:1 8:0 9:1 10:1 11:1 12:0 13:0 14:1 15:0 16:0 17:0 18:0 19:0 20:0 21:1 22:0 23:0 24:0 25:0 26:0 27:0 28:0 29:0 30:0 31:0 32:0 33:0 34:0 35:0 36:0 37:0 38:0  (before setup)
Board fingerprint is 4
Autoconfig: looks like TTGO v1 / Heltec v1/V2 board
Initializing SPIFFS
Reading initial configuration
configuration option 'led_pout'=9 
configuration option 'gps_txd'=-1 
configuration option 'maxsonde'=20 
configuration option 'debug'=0 
configuration option 'wifi'=3 
configuration option 'kisstnc.active'= 1 
Invalid config option 'kisstnc.a 

t
3
0
d

Init SSD1306 display 15 4
touch read 141: value is 1
Size of font list is 18
Display initialized
Showing welcome display
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
tou1
y
2 networks in networks.txt

RDZsonde: RDZsonde
DinoGast: Schokolade
before disp.initFromFile... layouts is 0x3ffbf20cReading from /screens.txt. available=5439
Line: '# Definition of display content and action behaviour'
Line: '#'
Line: '# Timer: (view timer, rx timer, norx timer)'
Line: '# - value -1: timer '
'
'
'
'
1
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
'
1
'
'
'
'
'
'
'
'
'
'
s
)
'
0
'
'
#
'
'
0
'
0
'
'
0
'
0
'
0
'
5
3
5
2
1
5
5
5
5
5
3
5
'
'
'
'
'
'
'
'
'
'
'
'
'
s
)
'
0
'
'
#
'
'
'
0
'
0
'
0
'
0
'
0
'
0
'
0
'
0
'
0
'
0
'
0
'
5
5
0
2
1
2
5
5
5
5
5
0
'
'
'
'
s
)
'
1
'
'
#
'
'
0
'
0
1
'
0
'
0
'
0
'
0
'
0
'
5
5
0
2
1
3
5
5
5
5
5
5
'
'
'
'
s
)
'
1
'
'
#
'
'
0
'
0
'
0
'
'
0
'
0
'
0
'
0
'
0
'
0
'
5
5
0
2
1
4
5
5
5
5
5
5
'
'
'
'
s
)
'
1
'
'
#
'
'
0
'
'
0
'
0
'
0
'
0
'
0
'
0
'
0
'
0
'
'
0
'
0
'
'
0
'
'
0
9
SX1278 configuration finished
Setup finished

Reading channel config:
Adding 402.299988 - 2 - 1 - Greifswald      
Adding 402.500000 - 2 - 0 - Schleswig       
Adding 402.700012 - 2 - 1 - HH-Sasel        
Adding 403.000000 - 2 - 0 - DeBilt          
Adding 404.100006 - 2 - 1 - Norderney       
Adding 404.299988 

1

1

Sonde::setup() on sonde index 0
RS41: setting RX frequency to 402300000.00
AFC BW: 12500.000000  RX BW: 6250.000000
Debug: bdd1=0, bdd2=button1 press (dbl:-1) (now:1): 0 at 0 (0)
Running main loop in state 2. free heap: 216508;
currentDisp:1 lastDisp:1
[WiFi-event] event: 0
WiFi interface ready
touch read 141: value is 1
[WiFi-event] event: 15
WiFi access point  stopped
[WiFi-event] event: 2
WiFi client started
[WiFi-event] event: 2
WiFi client started
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
touch read 141: value is 1
[WiFi-event] event: 1
```

dl9rdz, 2019-10-16:
Strange. I assume the board was working with an older software version? Pin selection for the display should be fine. I don't have such a board, but maybe some suggestions:

- One thing (the only thing I am aware of right now) that changed compared to previous version is the use of hardware I2C instead of software I2C. Should not make a difference, but you could try (around line 200 in Display.cpp) use the (commented-out) constructor for SW_I2C instead of the one for HW_I2C.  (The order of the arguments is different for the two constructors, so its not enough to simple replace "HW" with "SW)

- Another thing I have read about are power management features of your v2 board. You can try connecting GPIO21 to GND and check if the display then works.

istamov, 2019-10-16:
You are actually correct and it seems to be related to the power management feature!
Once I connected GPIO21 to GND before applying power, the display is now working :) Tested with versions "devel20191014", "devel20191014b" and "devel20191016."

istamov, 2019-10-19:
Just wanted to confirm that with the latest build (devel20191018-full.bin) everything is working with this board without having to connect GPIO21 to GND, so thank you very much for your great work! :)

closed 2019-10-19 by istamov