dl9rdz/rdz_ttgo_sonde #15: TFT orientation on OLED SSD1306

PreinfalkG, 2019-11-03:
Hi

**!!! 1.000 thanks for the great work !!!**

I have a short question about the configuration 'TFT orientation (0/1/2/3)'.
Should this also work with an OLED display SSD1306?
I have tested all the settings and unfortunately cannot see any change in the screen orientation ...

greetings günter

dl9rdz, 2019-11-03:
Hi, no, the orientation setting is only for the TFT display.
The u8x8 library used for the OLED display has no support for display rotation.

PreinfalkG, 2019-11-03:
Okay, thanks for the quick answer!

sp3rat, 2019-11-03:
Later, about 2019.06 I used for my OLED TTGO v1
command: "FlipMode"

u8x8->setFlipMode(2);

It was working.
Now, I do not try.

dl9rdz, 2019-11-03:
Ok, 180° flip can be added if there is some benefit to it. The TFT does support 90° rotations, that's not feasible on the OLED.

dl9rdz, 2019-11-10:
Latest devel can now flip the OLED by 180° (orientation=3, because this is 180° flip also for the TFT display). 

TFT orientation 0/2 (90° rotation) is not supported for the OLED display and will result in identical default orientation as with orientation=1.

closed 2019-11-10 by dl9rdz

PreinfalkG, 2019-11-11:
Hi dl9rdz

**T H A N K   Y O U**, the new OLED orientation works great!

Greetings
günter