dl9rdz/rdz_ttgo_sonde #18: No signal on t-beam

yeckel, 2019-11-12:
Hi,
After some update about a two weeks ago I'm not receiving anything on the T_BEAM_V07. The spectrum is all white (full columns) and later frequencies are toggling pretty fast too. On Helltec LoRa 32 with the same SW and config it works. A HW issue? But another T_BEAM(AXP192) shows the same behavior. Any idea? Simple LoRa RX/TX seems to be working. Changing the start frequency to 433 makes no difference.
Tnx, 73

yeckel, 2019-11-17:
The log output is 0,0,...0. However when I perform a scan in separate sketch using just the guts from the Scanner::scan() and SX1278FSK.h then I'm getting meaningful results.

yeckel, 2019-11-17:
It seems it stop working after calling initGPS() in setup()

yeckel, 2019-11-17:
Ok, solved. The gps_txd has to have some meaningful value. Like the correct 15 for T-Beam. Or -1 which is default. I had in settings 18 which is "LoRa chip select". That's explain it.

closed 2019-11-17 by yeckel

dl9rdz, 2019-11-18:
Ok, thanks für tracking this down.  I added a comment to the documentation that gps_txd should be set to -1. The value of gps_txd is passed to the Serial constructor, which will not configure the tx pin if set to -1.