dl9rdz/rdz_ttgo_sonde #19: last version is not working

ayasystems, 2019-11-24:
Last version labeled as update.ino in http://rdzsonde.mooo.com/ is not working please check it.

regards

dl9rdz, 2019-11-24:
This issue is not very useful without additional information. What is the base version you are updating? Do you mean the devel or the master update? 

Note that updates only work if the remaining parts of the image (flash layout, file system) are compatible. There was a recent modification in the flash layout, so you cannot update a pre-Nov2-image to a later version. This was necessary in order to have the binary images compatible with the images generated with the Arudino IDE. You also cannot "update" a post-Nov2 devel version to the latest (pre-Nov2) master version.

To make it easier to find out if an update is possible, more recent images have flash version information encoded into the file name. You should be able to update any running "-B1" version to the lastest devel version with devel/update.ino.bin because the latest devel version is also a "-B1" version.

bazjo, 2019-11-24:
sorry for being ranty, but https://youtu.be/53zkBvL4ZB4

edwin-oetelaar, 2019-11-24:
http://www.catb.org/~esr/faqs/smart-questions.html

more to the point ;-)

073
PA2LVD

dl9rdz, 2019-11-25:
closing this issue... feel free to re-open if you have a more detailed and reproducible problem description.

closed 2019-11-25 by dl9rdz