dl9rdz/rdz_ttgo_sonde #21: 2" TFT screen blank after change in screens.txt

DL2MF, 2020-02-02:
Hi Hansi,
today I did setup an additional new TTGO T-Beam from the latest series with 3 buttons. 

I also did wire a new TFT 2" display based on your wiring scheme and both brandnew units worked well from scratch. In my config.txt the display settings type= 1 and display screens 6,7,8,9,12,13 were configured.

After 30 mins operation I decided to make a minimal change to display Altitude instead of Distance  to screen 12 (Decode/General.TFT.Bazjo): 

before:
color=99004A
9,0=xDistance
color=FF007B
10,0=GD

after:
color=99004A
9,0=xAltitude
color=FF007B
10,0=A

After starting a ESP32 Sketch Data upload, the TTGO reboots, the screen resets and stays with no text and only white backlight. Uploads and compile are done via Arduino IDE, like I'm working since 1 year with TTGO. 

Investigating for any issues, I could not found a real reason, after 15-20 reboots, power off and flashing again, suddenly the display was back again in normal mode. Making the same change to screens.txt the error is reproducable and I'm going to take a coffee.

Cheers, Mike, DL2MF

dl9rdz, 2020-02-02:
Hi Mike,
thanks for the report. Unfortunately I am unable to reproduce the problem. I changed config.txt and screens.txt according to your description, and everything seems to work fine.

Maybe you can send me by email (a) the exact screens.txt that reproducibly causes the problem after a file system upload and (b) a copy of the serial console output after the upload?

Cheers, Hansi, dl9rdz

dl9rdz, 2020-12-28:
Closing this issue, as a lot has changed in the code since then, maybe the problem no longer exists anyway. Feel free to re-open the issue if there is a way to reproduce the problem.

closed 2020-12-28 by dl9rdz

DL2MF, 2021-01-06:
@dl9rdz 
Hi Hansi, 
just want to confirm using the latest version from 20201230 no current display issue exist.
tnx, vy 73
Mike