dl9rdz/rdz_ttgo_sonde #23: OLED by default on the wrong pin

MikeTango, 2020-06-24:
Moin,

just got the 1306 OLED to work on the TTGO, went through the defaults for board rev 1 first, the documentary says 4, 15 for pins SCL and SDA, but initializes 3, 15 (visible on console). Board rev is 1.0.

dl9rdz, 2020-06-24:
Can you explain in more detail what you think is wrong? If "1.0" refers to a TTGO or Heltec v1.0 (not a T-Beam v1.0), then by default (freshly flashed image) auto-selects SDA=4 SCL=15 as that is where the on-board OLED is connected the Heltec and TTGO v1.0 boards.

In that case you should see "Autoconfig: looks like TTGO v1 / Heltec v1/V2 board" and "Init SSD1306 display 15 4" on the console.

If that line reads "Init SSD1306 15 3", which is what your messages seems to indicate, the probably only way this can happen if you manually configured Pin3 in the config file. In all of the autoconfig default configurations , Pin 3 is never used at all.

MikeTango, 2020-06-24:
Typo, it initializes as follows when config.txt has disptype=0 uncommented and nothing else:

I2C device found at address 0x34 !
axp192 PMU found
I2C device found at address 0x3C !
ssd1306 display found
done

button1.pin is 38, attaching interrupt
disptype is 0
Display created
Init SSD1306 display 13 4
Display initialized
Showing welcome display

dl9rdz, 2020-06-24:
Ok, thanks for the clarification. So if you have a axp192 PMU,  you have a T-Beam 1.0. This board can be used with the large TFT or small OLED display. On first startup after flashing a full image, the software tries to guess which of the two is the case. In your case,  SDA=4 SCL=13 is the wiring for the large TFT; so apparently you first started the board without the OLED display correctly attached.  This is the expected behaviour.  

As soon as you manually modified the config, no autodetection will be performed any more. The initially detected configuration (or later manual modifications) will persistently be stored in config.txt. If you want autoconfig to happen again, flash a full image, or delete all display-related config items from config.txt and do a reset. Then the default values sda=21 scl=22 for OLED on T-Beam 1.0 should be selected as new default values.  But seems like you manually got it to work already anyway, so everything should be fine, right?  But everything is as expected....

MikeTango, 2020-07-23:
As written in email, i booted the board without a screen and it did exactly what it was supposed to do.

closed 2020-07-23 by MikeTango