dl9rdz/rdz_ttgo_sonde #24: Problems with Display OLED 1306

smarti2019, 2020-07-08:
Problems with OLED display 1306. When i press the buttons to reach the individual menu items, the display regularly crashes. So it is only confused characters are displayed.
Any Ideas?

smarti2019, 2020-07-08:
[image]

smarti2019, 2020-07-08:
[image]

smarti2019, 2020-07-08:
_

dl9rdz, 2020-07-08:
Can you check if this problem also happens if you use a touch pin as button instead of the power button?

smarti2019, 2020-07-08:
how do I implement a touch pin on the ttgo beam board?

smarti2019, 2020-07-08:
Ok i have found the Touch PIN. With this configuration and use of the touch pin, it works. But it would be nice if the existing buttons on the TTGO beam would work without a crash.
[image]

dl9rdz, 2020-11-11:
Can you test devel20201111b? It should fix the problem, but I misplaced my LCD display for the T-Beam to test myself.

smarti2019, 2020-11-11:
I Have just tested the latest devel20201111b version. The display problem has been resolved and the power button can be used as button 2. Absolutely perfect.
At this point, a big, big thank for you for this ingenious software!

closed 2020-11-11 by dl9rdz