dl9rdz/rdz_ttgo_sonde #25: Feature request: Decoder for IMET and RS14

MikeTango, 2020-07-23:
Hi, 

is it possible to add decoding support for these two?

greetings

MT

dl9rdz, 2020-07-23:
The difficulty with iMet is that it is AFSK modulated, which is not well-handled by the TTGO's FSK receiver chip. It might be possible to so some magic and decode strong signals, but it most certainly would perform poorly on weak signals.

Also, there are no imet launches nearby my location, and I don't have an imet for testing. An I/Q recording of an imet sonde could be useful for simulating a imet sonde with an sdr tx, but so far I have seen only recording of FM-demodulated signals.

I am not sure what sonde you are referring to with "RS14"? Never heard of a RS14, if you meant RS41, well, that is supported already. Or maybe PS15? That could be feasible, but I also lack material to test such decoder (and, to some extent, also the time to add such decoder to the software).

oeloeglioe, 2020-09-18:
m20?

dl9rdz, 2020-09-18:
m20 seems same modulation as m10, so should be possible; but I have no M20 to test with yet.

closed 2020-10-11 by MikeTango