dl9rdz/rdz_ttgo_sonde #31: precompiled binaries not available

gmke, 2020-09-10:
rdzsonde.mooo.com is not available

dl9rdz, 2020-09-10:
Thanks for the notice, the DNS entry has expired. Should be back in one hour.
(In any case, you can always access the github pages that back the web site at https://github.com/dl9rdz/rdz_ttgo_sonde/tree/gh-pages/devel and download the binaries there)

closed 2020-09-10 by dl9rdz

petanx, 2020-11-27:
Are the pre-compiled binaries defined for TTGO with 2 buttons or to the one with RST button only?
I have the TTGO with RST button and OLED display and the latest binary is not working. There is nothing on display, not sure if code is runnung. I also compiled version with SPI pins as they are on mine V2 board, but this compiled binary is not working either.

cixio, 2020-11-27:
> Are the pre-compiled binaries defined for TTGO with 2 buttons or to the one with RST button only?
Both

> There is nothing on display, not sure if code is runnung.
Please check if there is a wireless lan available called "RDZsonde". Connect to it with password RDZsonde and open http://192.168.4.1/

dl9rdz, 2020-11-27:
Binaries work on all supported boards, and in standard situations display pins will automatically be configured correctly (otherwise use the web interface to adjust). Make sure to flash the precompiled binary at offset 0x1000 (see picture in rdzsonde.mooo.com), otherwise it will not run.

If you compile the code yourself, make sure to not only upload the code, but also the file system data ("pio --target uploadfs" for platform io, "ESP32 Sketch Data Upload" on Arduino IDE).

In addition to scanning for the Wifi, you can also watch the serial port output when connected to a PC, the software is always very verbose on the USB port. For example Arduino console or "pio device monitor" for platform io.

petanx, 2020-11-28:
Thanks, my mistake was uploading the binary from 0x0000. Uploading from 0x1000 and works great.
I hope I will have some ideas or help you with this project in future, once I get used to it how it is working.