dl9rdz/rdz_ttgo_sonde #32: TTGO LoRa ESP32 With Only RST Button

dh47, 2020-09-29:
I got the model of TTGO LoRa that seems to only have the RST button, how do I go about adding one?

dl9rdz, 2020-09-29:
Option 1: Configure a "touch" input pin and use that pin as button by just touching the pin (or a piece of wire soldered to that pin). See, for example, https://microcontrollerslab.com/esp32-touch-sensor-button-example/, for a list of pins that can be used as touch input (of course only if not used by any other thing like connection to display). This might require some calibration of a touch pin threshold value (see Wiki for details). 
Option 2: Connect a real button to the board (between some pin and GND). Again, use a pin not used otherwise. For some pins (if I remember correctly that is GPIO >32, but I'm too lazy to verify right now), an additional pull-up register to VCC (3,3V) is required (value is not critical, something like 100K will do). For all other pints (<=32), the software will enable an ESP32-internal pull-up resistor.
Option 3: Connect your smartphone to the ESP32 via WIFI, access the web interface, there you can simulate button presses 😁

dh47, 2020-09-29:
I tired option 3 but its not showing up as a WiFi AP.  I could be doing something wrong, I'll link you to the purchase I made and you can see what I need to change.

[Banggood](https://www.banggood.com/LILYGO-TTGO-LoRa32-V2_1-ESP32-433MHz-or-868MHz-or-915MHz-OLED-0_96-Inch-SD-Card-bluetooth-WIFI-Wireless-Module-SMA-IP5306-p-1545144.html?utm_source=webpush&utm_medium=mail_webpush190_pc&utm_content=elaine&utm_campaign=newsletterwebpush_pps_copy&_branch_match_id=829605523939393573&ID=513922&cur_warehouse=CN)

dl9rdz, 2020-09-29:
The board is fine (one of the two I am using as well).

If the display shows something, the WiFi should also do something.

Did you upload the code but not the config files (data) using the Arduino IDE? If you either upload the full image (devel-xxx.bin) or upload first code and then file system data using Arduino IDE, you should have an active WiFi AP (unless you changed the config before uploading the file system data)

dh47, 2020-09-29:
I opened what I thought is the code needed to put onto the board RX_FSK.ino and flashed that on the board.

dl9rdz, 2020-09-29:
See https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/IDE-installation
Seems to me that you did not do what it says in the last line: "Upload data to SPIFFS with Tools -> ESP32 Sketch Data Upload"

dh47, 2020-09-29:
Oh, sorry about that, it did say to add that as esp32fs.jar to the tools folder.  I will try and do this now.

dh47, 2020-09-29:
That did it, thank you very much for the assistant.

closed 2020-09-29 by dh47