dl9rdz/rdz_ttgo_sonde #34: 1.3" OLED - Display not working in Spectrum Mode

df2ap, 2020-11-15:
I use RDZSonde on an TTGO T-Beam V1. The OLED-Display is an 1.3" OLED-Display SH1106. I flashed the BIN-File. Until Version 0.7.1 everythink working fine.

Now I flashed completly new with Version 0.8.0 Master. I made the right Configuration also new.

Everything works fine, but not the Spectrum Mode. There are only in the upper Part of the Display some Work.

[image]

cixio, 2020-11-15:
Same here, used: master_v0.8.0-B4-full.bin (Nov 11 07:51:45 2020)

dl9rdz, 2020-11-15:
Ah, thanks, I forgot about the SH1106 (which currently is treated the same way as the TFT, not the LCD, for configuring the spectrum layout)... update coming soon

dl9rdz, 2020-11-15:
Should be fixed in devel-20201115. (I don't have a SH1106 display to test)

df2ap, 2020-11-15:
It works! 

Thank you very much.

closed 2020-11-15 by df2ap