dl9rdz/rdz_ttgo_sonde #37: Changed screens.txt by error

puspis, 2020-11-21:
Hi. I am newbie to Github and have changed the screen.txt file by mistake. I have changed the original file for my own (modified). Sorry for the inconveniences. If you want, you can add my changes to the screens.txt file
[image]

dl9rdz, 2020-11-21:
You changed it in your fork, but its a nice alternative, will be glad to add it as an additional alternative in my next version.

puspis, 2020-11-21:
OK. See my screens.txt for other variations on the screens, ony for ILI9225 TFT.
Many thanks for your work. It's superb in combination with APRSdroid app.

closed 2020-11-28 by puspis