dl9rdz/rdz_ttgo_sonde #38: Air pressure and temperature from RS-41

digi2017, 2020-11-22:
Is it possible to display the temperature and Air pressure? That would be a very interesting additional info Screen.

dl9rdz, 2020-11-22:
Yes. If I have time (that's the tricky part :) as there are other more important items on my TODO list). Or if someone else sends a pull request.

Extracting temperature and humidity from RS41 data should be easy, open source GPL code is available both from Zilog and DXL. Regarding air pressure I am not sure, AFAIR DXL does just a calculation based on GPS height. But maybe that is fine for you, then it is a very easy part. Not all models of RS41 have a real pressure sensor anyway. RS41-SGP does, but I have not seen open source software extracting that value from the decoded data.

cygeus, 2020-12-27:
Thanks to the Ra project of DF9DQ an open source implementation to extract the air pressure from the decoded data of a RS41-SGP sonde is available @ https://github.com/einergehtnochrein/ra-firmware/blob/43dc11d06d1fe8893b047d8bf3f58fdf170743d8/src/rs41/rs41metrology.c#L76
Recently this got added to Zilog's repo as well: https://github.com/rs1729/RS/commit/5f13fd139a0c2c55846d2f9927f4addc602172dd

mycarda, 2021-02-09:
Added pull request https://github.com/dl9rdz/rdz_ttgo_sonde/pull/62. This calculates the temperature and humidity based on https://github.com/einergehtnochrein/ra-firmware and zilog80's code but my local RS41 radiosondes don't have pressure sensors so I did not add pressure. The values are not displayed. I leave that to somebody who know more about the display code.

dl9rdz, 2021-05-23:
Closing this issue as #62 has been merged

closed 2021-05-23 by dl9rdz