dl9rdz/rdz_ttgo_sonde #40: Boolean value config problem

analogic, 2020-11-25:
Hello, I am almost done implementing MQTT but I have one slight issue. See screenshot:

[image]

I am blind and unable to find problem - "MQTT active" should be boolean value or 1/0, but it still leaks some int no matter what I set up in form/config.txt/defaults. Source code changes are at https://github.com/dl9rdz/rdz_ttgo_sonde/compare/devel...analogic:devel

dl9rdz, 2020-11-25:
> On 2020-11-25, at 11:14, analogic <notifications@github.com> wrote:
> 
> 
> Hello, I am almost done implementing MQTT but I have one slight issue. See screenshot:
> 
>  <https://user-images.githubusercontent.com/934254/100213326-c371e100-2f0e-11eb-90ab-c73de37fa1c7.png>
> I am blind and unable to find problem - "MQTT active" should be boolean value or 1/0, but it still leaks some int no matter what I set up in form/config.txt/defaults. Source code changes are at devel...analogic:devel <https://github.com/dl9rdz/rdz_ttgo_sonde/compare/devel...analogic:devel>
> 

This line:
  {"mqtt.active", "MQTT Active (needs reboot)", 0, &sonde.config.mqtt.active},
assumes that active is an int (32 bit). In your case your a passing a (pointer to a ) boolean,
and interpreting this as a int pointer results in garbage...

(a) A simple solution is to use type "int" for mqtt.active
(b) Another solution is to add a new type to createConfigForm() for boolean. type 0 is an int and all positive numbers are strings with that length,
   so you could use any negative unused type number as "boolean". For example:
      case -7: // boolean
        addConfigBoolEntry(ptr, i, config_list[i].label, (bool *)config_list[i].data);
        break;
(with additional addConfigBoolEntry similar to addConfigNumEntry, but with bool instead of int) (and -7 instead of 0 in the line at the beginning.)

Cheers,
Hans

analogic, 2020-11-26:
Thanks a lot. I missed this line and active flag is working now as intended! Mqtt seems to be working without hickups now, but not yet tested with real packets - I don't have any original sonde available nor one flying around, I guess I'll just patiently wait before final PR

closed 2020-11-26 by analogic