dl9rdz/rdz_ttgo_sonde #41: Error compiling U8g2

lightswarm, 2020-11-29:
Hi,
I've just attempted a compile of the code and hit the following error:
    Using library U8g2 at version 2.27.6 in folder: C:\Users\David\Documents\Arduino\libraries\U8g2 
    exit status 1
    version.h: No such file or directory

Please advise

Here is the full compile results:

Arduino: 1.8.13 (Windows 8.1), TD: 1.53, Board: "TTGO LoRa32-OLED V1, 80MHz, 921600, None"

C:\Program Files (x86)\arduino-1.8.13\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\arduino-1.8.13\hardware -hardware C:\Users\David\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\arduino-1.8.13\tools-builder -tools C:\Program Files (x86)\arduino-1.8.13\hardware\tools\avr -tools C:\Users\David\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\arduino-1.8.13\libraries -libraries C:\Users\David\Documents\Arduino\libraries -fqbn=esp32:esp32:ttgo-lora32-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path C:\Users\David\AppData\Local\Temp\arduino_build_952135 -warnings=none -build-cache C:\Users\David\AppData\Local\Temp\arduino_cache_759751 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\David\Documents\Arduino\RX_FSK\RX_FSK.ino

C:\Program Files (x86)\arduino-1.8.13\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\arduino-1.8.13\hardware -hardware C:\Users\David\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\arduino-1.8.13\tools-builder -tools C:\Program Files (x86)\arduino-1.8.13\hardware\tools\avr -tools C:\Users\David\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\arduino-1.8.13\libraries -libraries C:\Users\David\Documents\Arduino\libraries -fqbn=esp32:esp32:ttgo-lora32-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path C:\Users\David\AppData\Local\Temp\arduino_build_952135 -warnings=none -build-cache C:\Users\David\AppData\Local\Temp\arduino_cache_759751 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\David\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\David\Documents\Arduino\RX_FSK\RX_FSK.ino

Using board 'ttgo-lora32-v1' from platform in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4

Using core 'esp32' from platform in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4

Detecting libraries used...

"C:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-IC:\\Users\\David\\AppData\\Local\\Arduino15\\ [... cut ...]
Using library MicroNMEA at version 2.0.3 in folder: C:\Users\David\Documents\Arduino\libraries\MicroNMEA 

Using library Ticker at version 1.1 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Ticker 

Using library SX1278FSK in folder: C:\Users\David\Documents\Arduino\libraries\SX1278FSK (legacy)

Using library SondeLib in folder: C:\Users\David\Documents\Arduino\libraries\SondeLib (legacy)

Using library U8g2 at version 2.27.6 in folder: C:\Users\David\Documents\Arduino\libraries\U8g2 

exit status 1

version.h: No such file or directory

dl9rdz, 2020-11-29:
This error is not related to U8g2, but to compiling RX_FSK.ino.  There should be a version.h in the same folder as the RX_FSK.ino, and with that it should compile fine without error.

lightswarm, 2020-11-30:
Hi,

 

Thanks for your help with that issue.  We have an additional compile error that is appearing:

 

exit status 1

'sondeTypeLongStr' was not declared in this scope

 

 

 

I have copied the full compile text below:

 

Looking forward to your reply.

 

David

 

C:\Users\David\Documents\Arduino\RX_FSK\RX_FSK.ino: In function 'const String sondeTypeSelect(int)':

 

RX_FSK:124:12: error: 'sondeTypeLongStr' was not declared in this scope

 

     sts += sondeTypeLongStr[i];

 

            ^

 

C:\Users\David\Documents\Arduino\RX_FSK\RX_FSK.ino: In function 'void addSondeStatus(char*, int)':

 

RX_FSK:364:102: error: 'sondeTypeLongStr' was not declared in this scope

 

   sprintf(ptr + strlen(ptr), "<tr><td id=\"sfreq\">%3.3f MHz, Type: %s</td><tr><td>ID: %s", s->freq, sondeTypeLongStr[s->type],

                                                                                                     ^

Multiple libraries were found for "WiFi.h"

Used: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi

Not used: C:\Program Files (x86)\arduino-1.8.13\libraries\WiFi

Using library AXP202X_Library at version 1.1.2 in folder: C:\Program Files (x86)\arduino-1.8.13\libraries\AXP202X_Library 

Using library Wire at version 1.0.1 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire 

Using library WiFi at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 

Using library ESPAsyncWebServer at version 1.2.3 in folder: C:\Program Files (x86)\arduino-1.8.13\libraries\ESPAsyncWebServer 

Using library FS at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 

Using library AsyncTCP at version 1.1.1 in folder: C:\Program Files (x86)\arduino-1.8.13\libraries\AsyncTCP 

Using library SPIFFS at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS 

Using library SPI at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI 

Using library Update at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Update 

Using library ESPmDNS at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS 

Using library MicroNMEA at version 2.0.3 in folder: C:\Users\David\Documents\Arduino\libraries\MicroNMEA 

Using library Ticker at version 1.1 in folder: C:\Users\David\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Ticker 

Using library SX1278FSK in folder: C:\Users\David\Documents\Arduino\libraries\SX1278FSK (legacy)

Using library SondeLib in folder: C:\Users\David\Documents\Arduino\libraries\SondeLib (legacy)

Using library U8g2 at version 2.27.6 in folder: C:\Users\David\Documents\Arduino\libraries\U8g2 

 

exit status 1

 

'sondeTypeLongStr' was not declared in this scope

dl9rdz, 2020-11-30:
Hi,

This looks like you updated RX_FSK.ino recently, but not libraries/SondeLib/Sonde.h

If you are not using the symlink approach (on MacOS or Linux), but instead copied the libraries folder to the
Arduino folder, and you did a git pull, then you have to copy the library folder again...

Chhers, H

lightswarm, 2020-11-30:
Thanks again for your help. Sorry to keep annoying you!

 

On Github under “Wireless Configuration” it says “The board will scan available WiFi networks, if the scan results contains a WiFi network configured with ID and Password in networks.txt, it will connect to that network in station mode.”

 

The networks.txt file only contains the following text:

RDZsonde

RDZsonde

DinoGast

Schokolade

 

Could you advise the format of the configuration parameters ?

 

David

dl9rdz, 2020-11-30:
Sure... Its two lines per entry, with SSID and Password for each entry.

First entry (first and second line):
First line: SSID for TTGO WiFi in Access Point mode
Second line: Password fo Access Point

After that: additional entries for APs the TTGO will try to connect to as a client device.
Each entry again with two lines for each network (first line SSID, second line password).

i.e. with the default configuration, it will connect to a single network with SSID "DinoGast" and Password "Schokolade",
or operate as an access point with SSID RDZsonde and password RDZsonde
-h

closed 2020-12-27 by dl9rdz