dl9rdz/rdz_ttgo_sonde #44: GPS is not working after FW installed

mmadela, 2020-12-23:
if it happens that the GPS is not sending any data however its LED flashes every 1s, here is a way to get T-Beam working:

compile, upload and run once
https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/examples/GPS/Example3_FactoryDefaultsviaSerial
then install the rdz_ttgo_sonde firmware again

the reason is a firmware being flashed at factory, GPS coordinates are displayed properly by this FW, but it apparently disables continuous sending of GPS data,
the rdz_ttgo_sonde does not send anything to GPS and does not alter existing GPS configuration

loefstrand, 2020-12-26:
The solution worked for me.

dl9rdz, 2020-12-27:
From what I can see, the linked code basically does a factory reset of the GPS chip. I have never heard any comments regarding GPS not working directly, but it would not be difficult to add such functionality to my software. Its just probing multiple baud rate in case the GPS is not on the default 9600 and sending the reset command....

smarti2019, 2020-12-30:
The GPS problem always came up when you had installed meshtastic, for example. Afterwards, GPS only works again after running the gps reset program.

loefstrand, 2020-12-30:
I had also tested meshtastic on my unit and after that the GPS didn't want to work until reseting it..

dl9rdz, 2020-12-30:
Ok, minor addition: If you configure the GPS TX pin (12 for the T-Beam 1.0). the software will send a "restore factory defaults" command to the GPS.  This should fix most configurations of the GPS chip done by other software.

Note that this will have effect only after you restart the GPS (i.e., power-cycle the T-Beam, pressing the reset button is not enough).

closed 2020-12-30 by dl9rdz