dl9rdz/rdz_ttgo_sonde #45: MQTT JSON error

loefstrand, 2020-12-26:
Hi!
I getting errors when receiving the MQTT messages the rdz_sonde_server publishes on my MQTT broker.
I have tested with both telegraf and Node-RED.

In Node-RED i get the following error message: `SyntaxError: Unexpected token } in JSON at position 563`
It's due to an extra comma sign before the } `countKT": -1,"crefKT": 3417,}`
If I remove the comma sign and inject the message in Node-RED, everything looks OK.

/// Marcus

dl9rdz, 2020-12-27:
MQTT has not really been tested by myself, it is provided by analogic's pull request (#42). Maybe he can help with this issue. Well, I can remove the comma, seems like it should not be there according to the JSON spec but is silently ignored by many parser. I don't have a ready MQTT environment for testing myself, so let me know if the latest commit resolved the issue (devel-20201227).

loefstrand, 2020-12-27:
I have updated my unit to devel-20201227 and waiting for the next weather balloon to see if it works...

loefstrand, 2020-12-30:
The data collected from the latest weather balloon, displayed on my Grafana server:

[image]

This is my flow of data: RDZsonde > MQTT broker > Telegraf > influxDB > Grafana

/// Marcus

dl9rdz, 2021-01-16:
Great, so this seems to be working fine now, closing this issue....

closed 2021-01-16 by dl9rdz

EricFROL, 2024-01-24:
Hello.
@loefstrand, do you still have that setup?

loefstrand, 2024-01-25:
Yes. I'm still using the same setup (now on devel20231110).
But "my" launch site is not that consistent at the moment, the last balloon launch was 20231221 :(

EricFROL, 2024-01-25:
Can you help me doing that same thing?