dl9rdz/rdz_ttgo_sonde #46: Not receiving any data

avdl72, 2020-12-27:
Hello,
I'm new to this kind of stuff and thought let's give a chance.

I ordered the 433MHz version of the TTGO LoRa32 board (https://a.aliexpress.com/_mrsU55L) and installed your software.
The instructions were very clear to follow to till this point.

After booting de TTGO I got a nice screen and was able to configure it via the web interface. Added my own WiFi and a nearby QRG.
But till today I wasn't able to receive any data :-(
Even when the sonde was flying above my head, and I was standing outside there was no data received.
I tried a lot by replacing it with another antenna, going back to the version 0.7.1 instead of v0.8.0_B4.
I don't know anymore ;-(

One thing for sure. Your software is working because I have already seen some good articles on the internet and video's on YouTube.
So even is this is properly not a software issue, is there a way I could get some assistance in getting my board working?

Thank you!

dl9rdz, 2020-12-27:
Difficult to diagnose remotely :-)

With a full-size vertical antenna outside, reception of RS41 sonde usually start working for me when they are about 80km away. With the small antenna that comes with the TTGO board inside a car its maybe 20..30km distance. Much less if the sonde already is very close the ground. M10 and DFM (new decoder) use a different software-based decoding approach, which might be slightly less ideal than the RS41 decoder, but I did not systematically check at what signal level / distance decoding becomes possible.

The RS41 decoder uses hardware AFC and should be reasonably robust against frequency error of the TTGO board. For DFM and M10 sondes, or if you have a really bad board in terms of frequency accuracy, manual calibration might help. Best done with a RS41 you receive on a known exact frequency. You can also try adjusting freqofs +/- 5Khz or +/- 10 kHz and see if this improves reception.

Note that the radiation pattern of the antenna has a minimum in the direction of the tip of the antenna. So if a sonde is right above you with its antenna hanging downwards, and you hold the board with the antenna pointing upwards, it might not be an ideal situation.  Inside a house it is pretty difficult to receive anything.

Apparently sometimes TTGO LoRa32 boards get mislabelled. I recently ordered two boards for an IoT project for 868 MHz, and what I got is one board labelled "868 MHz" and another one labelled 433/470 MHz, but the latter is not a 433/470 Mhz board (didn't try it on 868 MHz, but it is completely deaf on 433 MHz).

avdl72, 2020-12-27:
Thank you for the explanation.
I use De Bilt in The Netherlands which is a RS41 model to receive. This is the only one close to me.

How can I figure out if the board a 433MHz? Besides the label.

dl9rdz, 2020-12-27:
Regarding my mislabelled TTGO: You can remove the silver label, its a simple sticker. Below, a real 433 version has "HPD14A" engraved, my mis-labelled 868 MHz board with the 433 MHz label has "HPD13A" engraved below the sticker.

If yours is a HPD14A, the next best option is find someone nearby with a working TTGO. If that is receiving RS41 and yours is not, in identical conditions, than you likely have some form of hardware defect.

avdl72, 2020-12-27:
I have looked and mine is HPD14A.
I think I just order one more from another reseller.
And my other option is to drive to the launch site. 

Thanh you for your support.

closed 2020-12-27 by avdl72

avdl72, 2020-12-31:
My problem is solved.
I have ordered a new antenna and that worked perfect. I receive data now from a balloon about 60km away.

michaelegloff, 2021-11-01:
> My problem is solved. I have ordered a new antenna and that worked perfect. I receive data now from a balloon about 60km away.

hej there
which antenna did you order? I think I have a similar problem.