dl9rdz/rdz_ttgo_sonde #48: Feature Request: LMS6 radiosonde decoding

Shogoc, 2020-12-28:
I'd like to request the capability to decode LMS6 radiosondes. Most of the NOAA radiosondes launched in my area are LMS6's. This is a super exciting project because of the lower-cost hardware and I think that our radiosonde tracking footprint could be significantly increased if we could leverage the TTGO hardware.

bazjo, 2020-12-28:
As LMS6 are going to be phased out in the next 2-3 years and are to be replaced by RS41 and DFM-17, this feature would have an best before date right from its start

dl9rdz, 2020-12-30:
It is unlikely that I will implement such a decoder, but of course I will consider pull requests with working decoders for inclusion (if memory footprint is not too excessive, which I don't think).  Hardware-wise it should work (for the 400 MHz version), as modulation seems to be FSK.

kc0bqb, 2021-01-11:
I am curious if you have any more documentation for the SX1278FSK library.  I have started working on decoding the LMS6 radiosondes.  I am having some time understanding how to get the sx1278fsk library to recognize an lms6 packet.  I have changed the packet length (300) and sync (\x00\x58\xf3\x3f\xb8).

closed 2024-09-04 by dl9rdz