dl9rdz/rdz_ttgo_sonde #49: handlePMIirq() called. being logged repeatedly

eben80, 2020-12-29:
Hi,
I'm not sure what this means but I get `handlePMIirq() called. THIS SHOULD NOT HAPPEN w/o button2_axp set` being logged repeatedly in the serial logs. 

Is there something wrong or is this normal behavior?

I'm using TTGO V2.1 1.6 with GPS on 12 and the default touch buttons with calibrated values as per instructions.

dl9rdz, 2020-12-29:
What this means is that something is triggering interrupts on Pin 35. It should not happen, but it is not really a problem. (But I should disable interrupts in software if no AXP chip has been found).

What is shown in the "About" tab in the web interface? It should say "Fingerprint 31 (TTGO LORA32 v2.1_1.6)" with your board, but maybe with the the additional GPS it is Fingerprint 63?

eben80, 2020-12-29:
Ah OK I understand.

I've actually been getting both fingerprints 31 and 63.
My theory is that when it's connected on serial it gives me 63 and when not 31, that's what I've been seeing anyway.

In the meantime I've just commented out that warning and added some divider to better follow in console.
`______________________________________________________________________________

Running main loop in state 0 [currentDisp:0, lastDiso:0]. free heap: 106424;
RS41: setting RX frequency to 402100000.00
AFC BW: 12500.000000  RX BW: 6250.000000
TIMEOUT
RX result 1, laststate was -1
Timeout check: 85110 - 82647 vs -1; 85110 - 0 vs 0; 85110 - 85110 vs 0; lastState: 0
NORX timer expired
event b: action is 41
nextRxSonde: 0
receive(): Result is 8001 (action 128, res 1)
waitRXcomplete returning 8001 (RX_TIMEOUT)
Loop: triggering action Sonde=0 (128)
Sonde::updateState for event 128
Loop: action is 65, sonde index is 0
current main is 0, current rxtask is 0
updateDisplay started... rx task: activate=128  mainstate=0
rx task: calling sonde.setup()

Sonde::setup() on sonde index 0
updateDisplay done (after 76 ms)
Unused stack: 5768
______________________________________________________________________________`

So if I can safely ignore it, its all good.
I wanted to see if I'm able to save the decoded data with timestamps to the sd-card on this board but then I saw the touch pins are all used for the sd-card interface *facepalm*

Thank you for your help.

dl9rdz, 2020-12-30:
Its not a big issue, but in the most recent version, the message will be gone.
- fingerprint 61 also considered as TTGO LORA32 v2.1_1.6, and will not try APX chip probing
- IRQ handler only setup if APX button is configured as button in the config.

closed 2020-12-30 by dl9rdz