dl9rdz/rdz_ttgo_sonde #50: Best way to gain flash memory

eben80, 2020-12-29:
I would like to try and write the output that you get in the Data screen on the web interface to SDcard for when I'm outside with only the tracker.

I want to implement the SD_MMC library but it's already exceeding the available flash.

Is it possible to comment out some code, ie. fonts or display drivers that are not used to free up some flash memory?

Thank you.

dl9rdz, 2020-12-29:
For the main branch I intend to keep things as they are for now, but of course you can create your version with only the features you need. I can put "add #ifdefs to customize which features are included" on my TODO list, but not with high priority.... Probably you can remove those features with little effort:
- MQTT support
- RS92 support (relatively large because it includes a FTP client for fetching satellite data and a gzip decompressor for that)
- Display driver and fonts for one of the two display types (TFT or LCD) if you only use that. Probably will be quite invasive in Display.cpp

You can also change the flash layout to get more flash space. Currently it is OTA-enabled (basically this means that it uses twice the space, so you can run on version of the software and upload a newer image while the software is running, and then boot the new image).  Never done this myself, so you'll have to google for how to do it. Something like this: https://github.com/espressif/arduino-esp32/issues/703, just remove "app1" and make "app0" twice as big.

eben80, 2020-12-29:
Thank you very much for your help. I'll give it a go and see how far I get.

closed 2020-12-29 by eben80

eben80, 2020-12-30:
Thank you. I think I managed to use #ifdefs for RS92 and MQTT so now I managed to mount the SDCard successfully
`Sonde::setup() on sonde index 0
RS41: setting RX frequency to 400600000.00
AFC BW: 12500.000000  RX BW: 6250.000000
SD Card Type: SDHC
SD Card Size: 7624MB
Total space: 7618MB
Used space: 0MB`

Piter-NS, 2021-01-22:
> Thank you. I think I managed to use #ifdefs for RS92 and MQTT so now I managed to mount the SDCard successfully
> `Sonde::setup() on sonde index 0 RS41: setting RX frequency to 400600000.00 AFC BW: [...]

Can you share your version with SD Card support?

eben80, 2021-01-23:
> Can you share your version with SD Card support?

While I could mount the SD Card, I ran into issues with the coexistence of SPIFFS and SDCARD filesystems so I abandoned it. Probably the way to go would be to use only SDCard instead of SPIFFS but I didn't get around to that yet.
I was only able to mount the SDCARD in setup() but wasn't able to log anything yet. I might give it another go later. Bear in mind, if you use SDCARD, you lose most of the touch pins.