dl9rdz/rdz_ttgo_sonde #51: Almanac downloading issue

RegsJR, 2020-12-30:
Hi DL9RDZ,
it seems, there was something changed with FTP orbital data download, maybe the download URI? TTGO is successfully connected and tries to download the orbital data for some time (about 15 secs.), but fails. Where can be this bug? Thank you for any support, with the best regards, Milan

dl9rdz, 2020-12-30:
Hmm. Here everything seems fine, see below.

Maybe you can have a look at the USB serial output to better so what is going wrong for you.

```
WiFi connected
IP address: 
192.168.42.56
updateDisplayIP: 5 7
year 2020, day 365
now: 202036520, existing:  => updating
running geteph

220 Welcome to Geodesy FTP service.
331 Please specify the password.
230 Login successful.
200 Switching to Binary mode.
227 Entering Passive Mode (205,156,4,133,197,168).
205
156
4
133
197
168
connecting to www.ngs.noaa.gov:50600
fetching /cors/rinex/2020/365/brdc3650.20n.gz with FTP...
150 Opening BINARY mode data connection for /cors/rinex/2020/365/brdc3650.20n.gz (58611 bytes).
fetched 58611 bytes
Decompressing ephemeris data...

... (res=1) decompressed 8192 into 27060 bytes
written: 27060 of 27060
... (res=2) decompressed 1446 into 5708 bytes
written: 5708 of 5708
... decompressed 32768 bytes
... (res=1) decompressed 6746 into 29196 bytes
written: 29196 of 29196
... (res=2) decompressed 844 into 3572 bytes
written: 3572 of 3572
... decompressed 65536 bytes
... (res=1) decompressed 7348 into 31396 bytes
written: 31396 of 31396
... (res=2) decompressed 328 into 1372 bytes
written: 1372 of 1372
... decompressed 98304 bytes
... (res=2) decompressed 7576 into 32768 bytes
written: 32768 of 32768
... decompressed 131072 bytes
... (res=1) decompressed 288 into 1227 bytes
written: 1227 of 1227
... (res=2) decompressed 7395 into 31541 bytes
written: 31541 of 31541
... decompressed 163840 bytes
... (res=1) decompressed 797 into 3455 bytes
written: 3455 of 3455
... (res=2) decompressed 6833 into 29313 bytes
written: 29313 of 29313
... decompressed 196608 bytes
... (res=1) decompressed 1359 into 5635 bytes
written: 5635 of 5635
... (res=2) decompressed 6541 into 27133 bytes
written: 27133 of 27133
... decompressed 229376 bytes
... (res=1) decompressed 1651 into 6976 bytes
written: 6976 of 6976
... (res=0) decompressed 1237 into 5315 bytes
written: 5315 of 5315
done extracing content (total length: 241667)
Skipping header:      2.11           N: GPS NAV DATA                         RINEX VERSION / TYPE
Skipping header: teqc  2019Feb25     CORS-ADM Account    20201230 19:49:01UTCPGM / RUN BY / DATE
Skipping header: Linux 2.4.21-27.ELsmp|Opteron|gcc|Linux 64|=+               COMMENT
Skipping header:      2.10           N: GPS NAV DATA                         COMMENT
Skipping header: rvacn.e(1404.07)                        2020-12-30T19:27 UTCCOMMENT
Skipping header: This file contains the validated broadcast navigation       COMMENT
Skipping header: messages from the following files:                          COMMENT

```

dl9rdz, 2020-12-30:
Actually, there seems to be a small issue that sometimes causes a file corruption. This has been fixed in the most recent version.

RegsJR, 2020-12-30:
Thanks for your support! Some week ago was everything working well. I will try to reupload RXFSK. :-)

By the way, for the M2O option it seems, there is no value in the qrg table (just RS41, DFM, RS92). Should I do anything more for M2O decoding option?

dl9rdz, 2020-12-31:
If you use a firmware that implements M20 decoding (not really untested), e.g., devel20201229b, you should see the option for M20 in the web interface (qrg tab)...

RegsJR, 2020-12-31:
With the newest version it seems working well! Thank you very much, DL9RDZ :-)
'73 regs.jr

```
02:11:56.385 -> Connecting to: Auriga with password SA6lkcvOAT
02:11:56.613 -> touch read 13: value is 1
02:11:56.889 -> .touch read 13: value is 1
02:11:57.446 -> touch read 13: value is 0
02:11:57.493 -> touch read 13: value is 1
02:11:57.493 -> [WiFi-event] event: 4
02:11:57.493 -> Connected to access point
02:11:57.539 -> [WiFi-event] event: 7
02:11:57.539 -> Obtained IP address: 192.168.137.204
02:11:57.539 -> .
02:11:57.539 -> WiFi connected
02:11:57.539 -> IP address: 
02:11:57.539 -> 192.168.137.204
02:11:57.539 -> updateDisplayIP: 5 7
02:11:57.584 -> year 2020, day 366
02:11:57.677 -> now: 202036601, existing:  => updating
02:11:57.722 -> running geteph
02:11:57.722 -> 
02:11:57.815 -> touch read 13: value is 1
02:11:58.095 -> touch read 13: value is 1
02:11:58.419 -> touch read 13: value is 1
02:11:58.697 -> touch read 13: value is 1
02:11:59.019 -> touch read 13: value is 0
02:11:59.299 -> touch read 13: value is 1
02:11:59.620 -> touch read 13: value is 1
02:11:59.898 -> touch read 13: value is 1
02:12:00.222 -> touch read 13: value is 1
02:12:00.500 -> touch read 13: value is 1
02:12:00.823 -> touch read 13: value is 1
02:12:01.101 -> touch read 13: value is 0
02:12:01.425 -> touch read 13: value is 1
02:12:01.705 -> touch read 13: value is 1
02:12:01.987 -> touch read 13: value is 1
02:12:02.312 -> touch read 13: value is 1
02:12:02.589 -> touch read 13: value is 1
02:12:02.915 -> touch read 13: value is 0
02:12:03.193 -> touch read 13: value is 1
02:12:03.519 -> touch read 13: value is 1
02:12:03.799 -> touch read 13: value is 1
02:12:04.122 -> touch read 13: value is 1
02:12:04.397 -> touch read 13: value is 1
02:12:04.724 -> touch read 13: value is 0
02:12:05.002 -> touch read 13: value is 1
02:12:05.327 -> touch read 13: value is 1
02:12:05.606 -> touch read 13: value is 1
02:12:05.883 -> touch read 13: value is 1
02:12:06.208 -> touch read 13: value is 1
02:12:06.484 -> touch read 13: value is 0
02:12:06.809 -> touch read 13: value is 1
02:12:07.087 -> touch read 13: value is 1
02:12:07.388 -> touch read 13: value is 1
02:12:07.721 -> touch read 13: value is 1
02:12:07.999 -> touch read 13: value is 0
02:12:08.323 -> touch read 13: value is 1
02:12:08.603 -> touch read 13: value is 1
02:12:08.928 -> touch read 13: value is 1
02:12:09.203 -> touch read 13: value is 1
02:12:09.527 -> touch read 13: value is 0
02:12:09.808 -> touch read 13: value is 1
02:12:10.088 -> touch read 13: value is 1
02:12:10.412 -> touch read 13: value is 1
02:12:10.686 -> touch read 13: value is 1
02:12:11.011 -> touch read 13: value is 1
02:12:11.289 -> touch read 13: value is 0
02:12:11.613 -> touch read 13: value is 1
02:12:11.892 -> touch read 13: value is 1
02:12:12.216 -> touch read 13: value is 1
02:12:12.495 -> touch read 13: value is 1
02:12:12.821 -> touch read 13: value is 0
02:12:13.096 -> touch read 13: value is 1
02:12:13.421 -> touch read 13: value is 1
02:12:13.696 -> touch read 13: value is 1
02:12:14.019 -> touch read 13: value is 1
02:12:14.299 -> touch read 13: value is 0
02:12:14.623 -> touch read 13: value is 1
02:12:14.904 -> touch read 13: value is 1
02:12:15.183 -> touch read 13: value is 1
02:12:15.508 -> touch read 13: value is 1
02:12:15.788 -> touch read 13: value is 0
02:12:16.111 -> touch read 13: value is 1
02:12:16.388 -> touch read 13: value is 1
02:12:16.714 -> touch read 13: value is 1
02:12:16.992 -> FTP connection to www.ngs.noaa.gov failed
02:12:16.992 -> touch read 13: value is 1
02:12:17.084 -> Skipping header: 
02:12:17.084 -> reading RNX done, result is 0, ephs=0x0
02:12:17.318 -> touch read 13: value is 0
02:12:17.596 -> touch read 13: value is 1
02:12:17.919 -> touch read 13: value is 1
02:12:18.194 -> touch read 13: value is 1
02:12:18.514 -> touch read 13: value is 1
02:12:18.791 -> touch read 13: value is 0
02:12:19.116 -> touch read 13: value is 1
02:12:19.395 -> touch read 13: value is 1
02:12:19.720 -> touch read 13: value is 1
02:12:19.998 -> touch read 13: value is 1
02:12:20.090 -> SetupAsyncServer()
02:12:20.090 -> 
```

RegsJR, 2020-12-31:
With the newest version it seems working well! Thank you very much, DL9RDZ :-)
'73 regs.jr

```
02:11:56.385 -> Connecting to: Auriga with password ---
02:11:56.613 -> touch read 13: value is 1
02:11:56.889 -> .touch read 13: value is 1
02:11:57.446 -> touch read 13: value is 0
02:11:57.493 -> touch read 13: value is 1
02:11:57.493 -> [WiFi-event] event: 4
02:11:57.493 -> Connected to access point
02:11:57.539 -> [WiFi-event] event: 7
02:11:57.539 -> Obtained IP address: 192.168.137.---
02:11:57.539 -> .
02:11:57.539 -> WiFi connected
02:11:57.539 -> IP address: 
02:11:57.539 -> 192.168.137.204
02:11:57.539 -> updateDisplayIP: 5 7
02:11:57.584 -> year 2020, day 366
02:11:57.677 -> now: 202036601, existing:  => updating
02:11:57.722 -> running geteph
02:11:57.722 -> 
02:11:57.815 -> touch read 13: value is 1
02:11:58.095 -> touch read 13: value is 1
02:11:58.419 -> touch read 13: value is 1
02:11:58.697 -> touch read 13: value is 1
02:11:59.019 -> touch read 13: value is 0
02:11:59.299 -> touch read 13: value is 1
02:11:59.620 -> touch read 13: value is 1
02:11:59.898 -> touch read 13: value is 1
02:12:00.222 -> touch read 13: value is 1
02:12:00.500 -> touch read 13: value is 1
02:12:00.823 -> touch read 13: value is 1
02:12:01.101 -> touch read 13: value is 0
02:12:01.425 -> touch read 13: value is 1
02:12:01.705 -> touch read 13: value is 1
02:12:01.987 -> touch read 13: value is 1
02:12:02.312 -> touch read 13: value is 1
02:12:02.589 -> touch read 13: value is 1
02:12:02.915 -> touch read 13: value is 0
02:12:03.193 -> touch read 13: value is 1
02:12:03.519 -> touch read 13: value is 1
02:12:03.799 -> touch read 13: value is 1
02:12:04.122 -> touch read 13: value is 1
02:12:04.397 -> touch read 13: value is 1
02:12:04.724 -> touch read 13: value is 0
02:12:05.002 -> touch read 13: value is 1
02:12:05.327 -> touch read 13: value is 1
02:12:05.606 -> touch read 13: value is 1
02:12:05.883 -> touch read 13: value is 1
02:12:06.208 -> touch read 13: value is 1
02:12:06.484 -> touch read 13: value is 0
02:12:06.809 -> touch read 13: value is 1
02:12:07.087 -> touch read 13: value is 1
02:12:07.388 -> touch read 13: value is 1
02:12:07.721 -> touch read 13: value is 1
02:12:07.999 -> touch read 13: value is 0
02:12:08.323 -> touch read 13: value is 1
02:12:08.603 -> touch read 13: value is 1
02:12:08.928 -> touch read 13: value is 1
02:12:09.203 -> touch read 13: value is 1
02:12:09.527 -> touch read 13: value is 0
02:12:09.808 -> touch read 13: value is 1
02:12:10.088 -> touch read 13: value is 1
02:12:10.412 -> touch read 13: value is 1
02:12:10.686 -> touch read 13: value is 1
02:12:11.011 -> touch read 13: value is 1
02:12:11.289 -> touch read 13: value is 0
02:12:11.613 -> touch read 13: value is 1
02:12:11.892 -> touch read 13: value is 1
02:12:12.216 -> touch read 13: value is 1
02:12:12.495 -> touch read 13: value is 1
02:12:12.821 -> touch read 13: value is 0
02:12:13.096 -> touch read 13: value is 1
02:12:13.421 -> touch read 13: value is 1
02:12:13.696 -> touch read 13: value is 1
02:12:14.019 -> touch read 13: value is 1
02:12:14.299 -> touch read 13: value is 0
02:12:14.623 -> touch read 13: value is 1
02:12:14.904 -> touch read 13: value is 1
02:12:15.183 -> touch read 13: value is 1
02:12:15.508 -> touch read 13: value is 1
02:12:15.788 -> touch read 13: value is 0
02:12:16.111 -> touch read 13: value is 1
02:12:16.388 -> touch read 13: value is 1
02:12:16.714 -> touch read 13: value is 1
02:12:16.992 -> FTP connection to www.ngs.noaa.gov failed
02:12:16.992 -> touch read 13: value is 1
02:12:17.084 -> Skipping header: 
02:12:17.084 -> reading RNX done, result is 0, ephs=0x0
02:12:17.318 -> touch read 13: value is 0
02:12:17.596 -> touch read 13: value is 1
02:12:17.919 -> touch read 13: value is 1
02:12:18.194 -> touch read 13: value is 1
02:12:18.514 -> touch read 13: value is 1
02:12:18.791 -> touch read 13: value is 0
02:12:19.116 -> touch read 13: value is 1
02:12:19.395 -> touch read 13: value is 1
02:12:19.720 -> touch read 13: value is 1
02:12:19.998 -> touch read 13: value is 1
02:12:20.090 -> SetupAsyncServer()
02:12:20.090 -> 
```

dl9rdz, 2021-03-14:
Closing this issue as it seems to be fixed

closed 2021-03-14 by dl9rdz