dl9rdz/rdz_ttgo_sonde #53: APRS to radiosondy.info

ZL1LAC, 2021-01-14:
I notice theirs settings for APRS and to send to radiosondy.info  is this part of the software working?  I'm not seeing any aprs being sent over and nether is the owner of the website?  Would be cool to get this working as I'm wanting to setup a self contained station to track the balloons.

Regards,

dl9rdz, 2021-01-16:
The software supports encoding the data in APRS format. It works for aprsmap (via AXUDP) and APRSdroid (via TCP). The TCP protocol is the same as you would use for radiosondy.info.

I still have not yet implemented APRS authentication, which restricts it to local use.  Adding APRS authentication would also be easy, but so far I have intentionally not added it, because the decoder does not carefully filter out invalid decoded data.  This is fine for local use, sometimes it might even be better to see partially correct data (such as a DFM sonde with location but not yet fully decoded ID), but such data should not be sent to public servers.  So far I was too lazy / did not have enough spare time available for carefully checking the code such that it adequately flags if the data is fully correct and thus adequate for sending it to public servers.

ZL1LAC, 2021-01-16:
Bugger ok no worries, I was gonna use one of the ttgo boards to make a self contained radiosonde station to send to radiosondy.

closed 2021-01-17 by ZL1LAC