dl9rdz/rdz_ttgo_sonde #54: Various TFT display drivers

Piter-NS, 2021-01-20:
Is it possible to support TFT displays based on the ILI9488 or ILI9341 driver?

dl9rdz, 2021-01-20:
The software is rather modular and supports custom display layout definitions (screens.txt), so conceptually it is certainly possible, just adding a library and implementing another subclass of "RawDisplay" in Display.cpp is needed. But
- I have no plans of implementing it myself.
- An additional library will not fit into the ESP32 memory. So far, a single .bin is generated that contains all functionality for all configurations (different boards, different displays, different sonde types). For yet another display, you would have to generate/use a different bin image.

I guess someone with basic C skills can implement this on a weekend. If someone wants to implement this additional support, get in contact with me to discuss the organisation of conditional code integration (in case you want your contribution be merged in my repo).

Piter-NS, 2021-01-22:
OK, thanks for the tips

dl5dla, 2021-01-26:
Would be interesting if someone could add support for the ILI9341, as this display is available eqipped with a touch controller (XPT2046). So in general the Adafruit_ILI9341 library together with XPT2046_Touchscreen Library from Paul Stoffregen might be suited. But I guess there is a relevant effort even for the display part, because the currently used TFT22_ILI9225 library is quite different to the ILI9341 library.

dl9rdz, 2021-05-24:
I have replaced the TFT22_ILI9225 library with Arduino_GFX (https://github.com/moononournation/Arduino_GFX), due to strange issues with some (but not all) ILI9225 display with the previous library.

There is maybe not enough space to also include support for other displays in the standard binary image, but adjusting the code would now be as simple as replacing a single line in Display.cpp
(replace "tft = new Arduino_ILI9225" with "tft = new Arduino_ILI9341")

Probably you also want to adjust the display size (in getDispSize() in Display.cpp) and create a custom screen layout.

(Currently both old and new code is in the source, the #define ALT9225 in Display.h enables using the new library. The old library will eventually be removed)

dl9rdz, 2021-08-06:
ILI9341 support is now included (rudimentary... screen layout not yet optimized for making good use of the extended space, but good enough to see the same things as on the smaller ILI9225 display).

closed 2021-08-06 by dl9rdz

dl9rdz, 2021-08-06:
Actually, with my test display it is a bit more tricky. The default SPI speed is 40MHz, and this is outside the spec of the display, and occasionally causes crash of the display. There is now a config parameter to set the SPI speed. A value of 20 MHz (20000000) works stable for my display.  According to the data speed, it should be as low as about 10 MHz.