dl9rdz/rdz_ttgo_sonde #56: Board with only reset button, how to control the receiver ?

pjalocha, 2021-01-23:
Hello, writing for my friend: we managed to compile and upload the code, but the device has only one reset button, thus it is not possible to control it and the WiFi is not on, can you do something to turn the WIFi on or otherwise control the receiver ?
Thank you, Paweł.

dl9rdz, 2021-01-23:
By default the Wifi should be on. 

How did you upload the code? If you did not build a full image on your own, but uploaded the code alone, for example with Arduino IDE, make sure to also upload the file system as well.

You can use any touch input pin as additional button by configuring the pin as touch input (if you manually upload the config, modify the config in RX_FSK/data)

pjalocha, 2021-01-23:
Thank you for quick reply.
We uploaded the code with Arduino, how do you make sure the file system is uploaded as well ?

dl9rdz, 2021-01-23:
See https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Arduino-IDE-installation

Part 1: ESP32 Flash Filesystem Upload support => Install the Arduino IDE plugin
Part 2: Upload data to SPIFFS with Tools -> ESP32 Sketch Data Upload

pjalocha, 2021-01-23:
Thank you, it worked after all !

closed 2021-03-14 by dl9rdz