dl9rdz/rdz_ttgo_sonde #57: Is the LILYGO® TTGO T-Beam V1.1 ESP32 supported?

DL5UH, 2021-01-23:
I wonder if the version 1.1 is supported. And, should the LED of the GPS light up?

dl9rdz, 2021-01-23:
I don't have a V1.1 board to test, so short answer: I don't know.

Recently a few people had problems with the T-Beam and GPS, because the GPS chip was not in its factory default configuration (see, for example, https://github.com/dl9rdz/rdz_ttgo_sonde/issues/44), i.e. either operating at a different baud setting or configured to not send NMEA data automatically. I don't know if these were all V1.0 or also some 1.1.

If this is the cause of your problem, you can fix that by manually configuring the GPS TX pin (set it to 12 on the T-Team 1.0, probably the same on the T-Team 1.1?)

dl5dla, 2021-01-23:
I can confirm that V1.1 is working, including GPS data. To troubleshoot, I would recommend to first check, whether the GPS module on the board sends the GPS data  to the ESP32. This can be done with a simple sketch which reads the GPS data and outputs it on the serial console.

If the red LED is blinking (=GPS fix exists) but no GPS data is read, a factory reset (see link above) might help.

DL5UH, 2021-01-24:
GPS TX = 12 did not help, but the factory reset  [#44](https://github.com/dl9rdz/rdz_ttgo_sonde/issues/44) did. Thanks.

dl9rdz, 2021-01-24:
"did not help": Did you do the full procedure (Set GPS TX=12, then restart the TTGO (for example using the reset button) and let it complete the startup, then power off the GPS chip (best done by removing both USB cable and battery), then power it on again)?

DL5UH, 2021-01-25:
I had thought about removing the GPS battery, but unsoldering it without short circuit seemed too risky.

dl9rdz, 2021-01-25:
NOT the tiny GPS battery. I of course meant the normal 18650 battery!

What my software does is send a command to the GPS to restore the persistent configuration to factory defaults. This persistent configuration will be used by the GPS chip after it does an internal reset. A GPS reset is not caused by pressing the on-board reset button (this resets the ESP32, not the GPS).  So my software should work if you
(1) Configure GPS TXD
(2) Restart the ESP32 using the reset button (the GPS TXD pin configuration triggers the GPS factory reset only on startup) (Requires a recent enough devel software version)
(3) After that restart the GPS chip. This is NOT happening if you press the Reset button, but if the main power is removed (i.e. both USB cable and the "big" battery -- NOT unsoldering the tiny GPS backup battery). Probably, a long press on the power button should also work, but I did not test that.

I was just curious if the "did not help" was caused by you doing only step (1) and (2) but not (3), or if your GPS chip was in a state where my reset procedure was not sufficient.

DL5UH, 2021-01-26:
I did the test after configuring GPS TXD with power down (long pressing) and waiting. Than, I started the unit outdoor to get the GPS confirm LED. But I did not take the 18650 out.

Herfried, 2021-02-06:
My LILYGO® TTGO T-Beam V1.1 ESP32 is full funktional. 
But M20 sonde is not ´decoding. Why?

dl9rdz, 2021-02-06:
Sorry, cannot help you there. I have no M20 around to test myself. I have one confirmation that the latest firmware works fine for decoding the M20 from Vienna, and no other reports so far that something is not working.

Herfried, 2021-02-06:
Thanks for your quick responde. I live south of Vienna and I try to decode M20 sondes. I can do it good with SondeMonitor, MySondy GO and with http://www.om3bc.com/eng.htm , for Raspy (all last versions). I think it is an hard job to create a software without a M20 sonde. Is it a help for when I send you a MP3 record from a M20?
But I can say your rdzTTGOsonde software is the best and I like to use it !!!
Regards, Herfried

dl9rdz, 2021-02-06:
I assume you are using a recent devel version of the software.

What you might have to do is to modify the "M10/M20 RX bandwidth" setting, I guess the current default value of 12500 is too low, and you can try setting it to at least 19000.

A mp3 recording will not help much, its fine for PC software that decodes an audio stream, but for the TTGO receiver you need a real RF signal.

Herfried, 2021-02-06:
Thanke you. 
I will try it next time when a M20 is launched in Vienna.

Herfried, 2021-02-07:
Today an M20 is in Vienna active. I modified "M10/20 RX bandwidth" setting to 20800 and 15600. But without decoding M20.

dl9rdz, 2021-03-14:
Previous version had a bug with the GPS reset, which is fixed in the most recent devel version. I have tested it with first running Meshtastic software (which reconfigures GPS to not sending NMEA messages) and then running rdzTTGOsonde software, and it activates GPS reliably. So the software **should** work now out of the box on those T-Beam 1.1 which come with Meshtastic installed by the vendor.

(No news regarding M20, some people say it works others report it does not, so at least under some conditions it should be ok. I don't have M20 to test myself, so feel free to report positive and negative results to me).

Herfried, 2021-03-15:
Hallo. Regarding M20. 
Days ago I hunted 2 Vienna M20 in good condition. I use your latest software for my T-Beam 1.0, 1.1. and "TTGO LORA v2.1_1.6".
By scanning all preinstalled sonde QRG´s, the scanning mode never stopps for example at 401.003 MHz M20 or decode it. On the config page I can only adjust "M10/M20 RX bandwidth" and "M10/M20 AGC bandwidth" ... is it so???
By using the software SondeMonitor I must change "frame lenght to 45 (value in hex) for decoding M20 !
Default the value in hex for M10 is 64 ! .... for example. 

best regards,
Herfried (herex)

dl9rdz, 2021-03-15:
Do you use the "spectrum display" mode on your devices?

If yes: The spectrum sets a fast frequency hop mode on the SX1278, which seems to cause problem with decoding M10 (and likely M20 as well). devel200315 resets the fast hop mode when decoding M10/M20, so you might try if that solves your problem.

Yes, the only configuration option you can change are the two bandwidth values.  M10 and M20 have the same preamble, and after that the first two bytes are used by the software to distinguish between M20 and M10, and adjust the frame length automatically.

Herfried, 2021-03-16:
Hallo. Yes I use "spectrum display". After installing your newest software the scan stops and shows M20 datas without changes in config by remaining defaults. 
But important I will test all my boards in live conditions by receifing an Vienna M20. I will not activate one of my huntet M20 sondes, before I reduce TX power to prevent "errors" on this FQ, you understand. 
I will informe you if I have results. Thanke you.

Herfried, 2021-03-16:
Hallo DL9RDZ.
Today I tested all my 3 boards "T-Beam 1.0, 1.1. and "TTGO LORA v2.1_1.6"during an Vienna M20 live mission. 
Decoding is now nice and ok.

"M10/M20 AGC bandwidth" setting is in my case 7800.
"M10/M20 RX bandwidth" setting is in my case 15600.
All indications are "normal", but only Sonde Serial is indicated only as "M".
Scanning stopps from time to time on an "M10" stored FQ but indicates correct M20

Thanke you. Now is your very nice software M20 ready.
In future time I will test it during hunting on field.

best regards, Herfried (herex)

dl9rdz, 2021-03-16:
Thanks for testing, great to hear that decoding is working.

I have not yet implemented serial number decoding, as it is not yet fully clear how to do this, and was waiting for some clarification.
The serial number printed on the sonde is something like " ABB-C-DDEEEE", with claims that "C" identifies the production site.
oe5dxl uses some formula for calculating "C" that only produces values 1 or 2
zilog uses a different formula that yields only values >= 2
There have been reports for sondes with C values of 1, 2 and 3, so neither of the two formulas would work for all.

I can for now add serial decoding the way it works in dxl in the next version, which is fine for Vienna and most other places that current use C=2 sondes.

Herfried, 2021-03-17:
Only for your information. 
My hunted sondes SN:  
002 2 01536 = ME8101536 and 
002 2 01533 = ME8101533
... both from Vienna

73-de-LZ, 2022-02-08:
Great, i am using LILYGO T-Beam T22_V1.1 and i can confirm it is working loaded with the latest DEV release, i have an GPS issue not after unboxing, but after few experiments. Flashing https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/firmware/GPS_%20reset/GPS_%20reset.bin help me. I am using battery from Iphone 5, soldered and glued on the back of PCB, it is hard to just disconnect it, so GPS TX = 12 (RX on IO34), short of GPS backup battery and power cycle does not help me. 

[image]
[image]
[image]
[image]
[image]

Hraph, 2023-04-21:
@73-de-LZ Thanks for this inspiring mod. How did you safely connect this lipo battery instead of a 18650 one to the board?

73-de-LZ, 2023-05-01:
> @73-de-LZ Thanks for this inspiring mod. How did you safely connect this lipo battery instead of a 18650 one to the board?
Well excuse me, that will go offtopic, the answer is: Hard
First be sure that the battery is fully discharged, then desolder the connector from the battery flex cable, then find where is plus and minus on the cable, it's double side FPCB, find where you can cut it and remove the black coating, so to be able to solder it on the TTGO PCB plus pad. For the minus you will have to scratch the TTGO GND near Vbat pad. Finally apply original  double side adhesive tape as the one which Apple start to using in all their phones. In fact i doesn't like these "garage" solutions, but they are widely used now in the cellphone industry, almost in any smartphone now, as it is easy and fast to be produced. This tape you can remove anytime with applying few drops of alcohol and pull it slowly in the perpendicular direction to the adhesive side. And while we are offtopic just to mention, the GPS antenna that came with TTGO as on the photo is really bad, after upgrading it with a good one, square and bigger, with a good LNA inside, GPS working far better. It is again glued with double adhesive tape on the u-blox module.

dl9rdz, 2024-09-04:
As the T-Beam 1.1 (and 1.2) are fully supported for some time, it is time to close this issue :)

closed 2024-09-04 by dl9rdz