dl9rdz/rdz_ttgo_sonde #58: MQTT and Home Assistant

sebikolo, 2021-01-24:
How to configure MQTTo receive data by Home Assistant? I have a Mosquitto broker.

Data feed configuration - RDZSonde Server

MQTT Active (needs reboot) - 1
MQTT client ID - core_sonde_server
MQTT server hostname - 192.168.1.90 (Home Assistant IP)
MQTT Port - 1883
MQTT Username - homeassistant (user Mosquitto broker)
MQTT Password - sd66aas ..... (password Mosquitto broker)
MQTT Prefix - nazwa_sonde_server /

In Home Assistant, I cannot see data from RDZSonde Server.

ZL1LAC, 2021-09-05:
Did you solve this?

mycarda, 2021-09-21:
@sebikolo 

> MQTT Prefix - nazwa_sonde_server /

I use MQTT and I can see the messages in home assistant
[image]
 I did notice you have a space between your server name and the slash at the end. I think it's OK to use a space in the name and it works with my setup but I think it is unusual. I have no user and password for my mqtt test broker but other than that, I think our configurations are pretty similar.

What happens when you listen to topic `#` ? This should show everything.