dl9rdz/rdz_ttgo_sonde #59: Garbled display ILI9225 on latest version

mickr1969, 2021-01-31:
Hi,
I seem to be having an issue where some of the display is displaying garbage rather than the correct text.
master_v0.8.0-B4-full.bin version seems to display correctly, but if I load devel20210124-B8-full.bin
On it looks like the following

[image]
[image]
[image]

Just wondered if you have any ideas.
Kind Regards
Mick

dl9rdz, 2021-01-31:
This looks like you have the TFT configured for portrait format ("TFT orientation"), but you are using a display layout ("screen config") for landscape format. The display driver creates garbage if the displayed text exceeds the right-hand display limit.

Set "screen config" to 3, or configure the TFT for landscape mode (should be orientation 1 or 3 AFAIR)

mickr1969, 2021-02-01:
Hi, thanks for that. Yes you are correct. I had the screen configured to type 2.
Screen config (0=old, 1=OLED, 2=TFT, 3=TFT[port])
I thought type 3 was something to do with the control of the screen, as in I/O port rather than "portrait" mode.

Thanks for clarifying.
Kind Regards
Mick

closed 2021-02-01 by mickr1969