dl9rdz/rdz_ttgo_sonde #61: Idea: simulation mode

lastv8, 2021-02-06:
What about a simulation mode for the t-beam? 
There are locations where you do not have sonde landings so often. There might be people who would like to do some trials with their t-beam's. A second used case would be cold-sonde searches. People that like geocaching could also use this feature. 
I mean-  for sure - having a smart phone would be sufficient for example 2 and 3.
This idea would require a possibilty to input the location at the web interface and a software switch to start the simulation mode.

DL2MF, 2021-02-22:
Memory in TTGO is limited, is it really a benefit to migrate a reliable field tool into a complicated multi-purpose device? 
If you want trajectory simulations why not generate your own tracks from predictions or record a sonde and feed the IQ data / audio file into your favorite decoder?

lastv8, 2021-02-24:
I fully agree with your statement. Memory should not be wasted. Better to save it for further sonde protocols or display features.