dl9rdz/rdz_ttgo_sonde #64: Graw DFM sonde decoding of longitude not plausible in North America

WH3080, 2021-02-19:
Dear DL9RDZ and enthhsiasts. I am using the fantastic software you created to track weather baloons in North America. I installed it on a V1.1 T-Beam. The latitude received from Graw radio sondes (with the DFM 9 old or DFM setting) appears to be accurately decoded in the 53s N, whereas the longitude is not. The system always shows longitudes in the 300s. Longitude however should be displayed in -110s W range. I assume there might be a flaw in the deoding software or did I miss a setting? Any help is greatly appreciated. Thank you!

dl9rdz, 2021-02-20:
I can confirm that negative latitudes and longitudes are not handled correctly for DFM. Can you try devel20210220 and verify if that problem is fixed?

closed 2021-02-20 by dl9rdz

WH3080, 2021-02-20:
Works like a charm now. Thank you for the quick response DL9RDZ!