dl9rdz/rdz_ttgo_sonde #65: RS92 decoding does not work

kmp56, 2021-03-02:
RS92 decoding does not work. Only the radiosonde ID and data packets are shown. No coordinates and no other data

dl9rdz, 2021-03-02:
Your TTGO did successfully download ephemeris data?

kmp56, 2021-03-02:
Yes. I will check on another RS92, today's Lindenberg just lands and I don't hear it anymore ...

dl9rdz, 2021-03-02:
Since Linz and Hohenpeissenberg stopped using RS92, I did not really test the RS92 decoder any more, but I just put my old test RS92 outside the house and for that it seems to be working fine. It took a while to get 3 sats in view for the RS92 on ground for a 2D fix, but it decoded the position. Frame#, number of sats and GPS time should be shown on the web interface even if the RS92 does not have sufficient sats for a position calculation.

[image]

kmp56, 2021-03-02:
OK. Thanks for now, I'll check on the archival RS92 over the weekend. I will let you know the results ...

om1973, 2021-09-14:
I just tried to decode RS92 data using the igs.bkg.bund.de ftp server, however, no position was decoded. 4 satellites were visible. Is there any way to check whether the TTGO successfully downloaded the RINEX data? Also, is there any special WIFI setting required for that (I am using  station mode). PS: zilog pipeline decoding of the same RS92 with the latest RINEX file from the server above worked flawlessly.

dl9rdz, 2021-09-14:
- It download RINEX data only after a reset, and only if a RS92 entry is active. So you have to (a) activate a frequency entry with sonde type RS92, then do a reset
- The download will be shown on screen (You should see multiple lines: the ftp host, the file name, the size, "uncompressing" and ok/uncompressed size.). That's the only way how to see currently if the download succeeded. I can add a RINEX status in the web interface tough. If it completely fails, it will retry after the next reboot.
- I have not tested it much recently. The decoder is quite memory hungry, if the board crashes during decoding then it is some kind of out-of-memory problem.

om1973, 2021-09-14:
Thank you! Just tried it out - but no success. I have one RS92 entry active, but after reset no info about the RINEX download is shown on the screen. The web interface shows the correct ID and frame#, however the number of sats is 0 and the GPS time is out of date. Simultaneous decoding with zilog still works. I am still using version 20210213. Could this be an issue?

om1973, 2021-09-14:
PS: The same behaviour with version 20210402 running on another TTGO .

om1973, 2021-09-14:
Wifi mode 3 did work! Thank you :) I saw the RINEX status messages flashing up correctly this time and decoding started. However, then after about 5 sec both boards (T-BEAM V1.1 and TTGO V2.1) crashed during the RS92-decode and the boards reset. This is probably the out-of-memory issue you mentioned before. Anything one can do about it?

dl9rdz, 2021-09-14:
Ok, so a fix for the stack overflow / reboot will be on the next version

om1973, 2021-09-14:
That would be great! Thanks a lot for looking into it!

dl9rdz, 2021-09-23:
Tested, seems to be working fine now.
Make sure to have WIFi mode 3 and reboot after activating a RS92 in order to trigger eph download.
Also make sure to use the bund.de server, noaa.gov does not offer ftp download any more.

closed 2021-09-23 by dl9rdz