dl9rdz/rdz_ttgo_sonde #66: Chasemaper data feed

MCmakerPro, 2021-03-04:
I was wondering if its possible to feed data to a device running chasemaper via Payload Summary UDP messages or any other method.

LukePrior, 2021-05-25:
While not chasemapper you can now upload to SondeHub

MCmakerPro, 2021-05-25:
Hello, thanks for the update. I have been following the sondehub pull request for a while now. Thanks for the contribution!

darksidelemm, 2021-06-02:
I need to look into the UDP packets that are sent via this software... it may be possible to make chasemapper support them.

darksidelemm, 2021-08-04:
Sorry to barge into a thread, but @5B4APV  could you possibly email me at vk5qi (at) rfhead.net ? I have some questions about your auto_rx configuration.

fkemps, 2021-08-11:
I tried the UDP feed from latest rdz_ttgo_sonde (devel20210811-B14-full.bin) into Chasemapper, not been recognized yet.
This would be a welcome integration and looking forward to it.

Happy to give it a test run.

darksidelemm, 2021-08-11:
The AXUDP feed is a different to format than what Chasemapper supports, but it's probably going to be easier to just add support for AXUDP into chasemapper than adding the chasemapper format into this repo. Unsure when I'll get to it though.

darksidelemm, 2021-09-17:
Looking more into this, the AXUDP feed is not what I thought it was (some kind of verbose-ish format, instead it's APRS packets :-/ ). I don't really want to have to deal with APRS or KISS decoding in chasemapper.

The JSON format that is uploaded to Sondehub is 90% of the way to what chasemapper accepts as input via UDP, so maybe that's a way forward...

dl9rdz, 2021-09-21:
Ok, no problem, not a big deal to put some json structure into a UDP message.  Should be in most recent devel today. Just lightly tested, so don't expect it to be complete or bug-free, but its a starting point....

dl9rdz, 2021-09-22:
Ok, the real reason for that was slightly different to what I initially expected. Actually the sondehub network code was waiting for a newline character at the end of the response that sondehub did not send, so we blocked the main loop thread for a second (the default network timeout). This is why the decoder had already overwritten the old data with the new data, which we then sent twice.
Should be fixed now.

dl9rdz, 2021-09-30:
OK, this is working nicely, so let's close the issue.

closed 2021-09-30 by dl9rdz