dl9rdz/rdz_ttgo_sonde #67: Buzzer for RDZS

RegsJR, 2021-03-15:
Hello DL9RDZ, please, would it be possible to add support for a piezzo passive buzzer on TBeam and wifi-only TTGO? Make the buzzer beep when is the module able to decode the sonde?

dl9rdz, 2021-03-15:
Currently you can use the LED pin with an active buzzer. Using it with a passive buzzer would just require changing a few lines of code in flashLed() in RX_FSK.cpp, maybe with a configuration option. I'll keep this idea on the list of possible future enhancements.

RegsJR, 2021-03-16:
Thank you very much for support! I will try. So, LED pin and the ground pin, am I right?

dl9rdz, 2021-03-16:
Yes. The pin configured as LED pin is pulled HIGH for 700ms (within 1s) if a correct frame was decoded, and for 100ms if the start of a frame was detected, but not correctly decoded (CRC error for most types of sondes).

gariac, 2021-07-19:
I don't see a LED pin on the ttbeam 1.1.  Is there an equivalent?

73-de-LZ, 2022-02-08:
It is by default on IO4, the red led near blue charging led, it falls under the OLED usually and it is connected to Vcc 3.3V instead to ground, so my advice is to add external LED, for example on port IO25 trough 2K to GND and configure it on "Hardware configuration"/"LED output port" = 25. 

As for the buzzer it is easy to add an active one as Hansi advice (active who generates fixed freq. sound when powered) with micro switch, or you can switch it off by LED output port" = -1, but this will switch off the LED also.