dl9rdz/rdz_ttgo_sonde #68: No Data from TTGO in Android App rdzSondyGO

smarti2019, 2021-03-16:
With the current version, data from a received radiosonde is no longer displayed in the Android app. However, the connection to the TTGO is established. This Problem exist since Version from 14 and 15 March.

dl9rdz, 2021-03-16:
Thanks. fixed.

closed 2021-03-16 by dl9rdz

LukePrior, 2021-04-01:
Where can I find this app?

LukePrior, 2021-04-01:
I downloaded v1.0.5 but I'm not sure how to connect, both devices are on the same network is AP mode required?

vanThijs, 2021-04-06:
I have the same issue with v1.0.5 and the rdzSondyGo app: No connection between the App and the TTGO.

LukePrior, 2021-04-06:
> I have the same issue with v1.0.5 and the rdzSondyGo app: No connection between the App and the TTGO.

The app started working for me once the device had detected a Sonde.

vanThijs, 2021-04-06:
And how was it configured: was the ttgo in AP mode, or connected to an infrastructure WiFi network?

LukePrior, 2021-04-06:
> And how was it configured: was the ttgo in AP mode, or connected to an infrastructure WiFi network?

The two devices were both connected to the same network.

dl9rdz, 2021-04-06:
It has to be the same network.
Either TTGO in AP mode and smartphone connected to that AP (tricky if you want internet access for the map at the same time)
Or TTGO and smartphone connected to the same external access point (without firewall. some "guest" networks of access points limit IP traffic)
Or Smartphone as AP (tethering) and TTGO connected to the Smartphone AP (probably the most suitable configuration for mobile use)
All variants should work.

vanThijs, 2021-04-06:
Thank you for the clarification!

pavolgaj, 2022-04-13:
I have the same issue with the latest development firmware and Android app as mentioned by @smarti2019. App shows TTGO as connected (green icon), but no data received from sonde is displayed.
The Master version of rdzTTGO works fine.

dl9rdz, 2022-04-13:
Ok, thanks for reporting this.

The latest devel version has some changes to the JSON exporter code, this might be the cause of this problem. Will have a look at it the next days...

dl9rdz, 2022-04-14:
Should be fixed in the  update  today

pavolgaj, 2022-04-14:
It still doesn't work. App (in info window) shows green circle = data from the radiosonde. But, displayed data are default values of the app (e.g. sonde id is A123456).

dl9rdz, 2022-04-14:
Which version of the app are you using? 
I was testing with v1.0.6, and this seemed to be working fine here.

pavolgaj, 2022-04-14:
I have also v1.0.6. And tested with standard RS41 sonde.

dl9rdz, 2022-04-14:
Hmm. That is strange. That is exactly my test setup here. devel20220414 , app 1.0.6, and a RS41. 

In case you have Android tools installed, you could put your phone in USB debug mode, run "adb logcat" and send me the output

pavolgaj, 2022-04-14:
I found the reason for my problem. I test it inside - therefore sonde hasn't GPS data. And probably TTGO doesn't send a JSON string for incomplete data.
When I put sonde outside and got the position, App shows the correct information.

In the previous version of firmware, I tested it always inside without a similar problem. You maybe implemented some additional filtering for uploading data to App.