dl9rdz/rdz_ttgo_sonde #69: Errors on LCD

sebikolo, 2021-03-24:
LCD does not display information correctly. How to fix it?

https://drive.google.com/file/d/1YP0_DvDjzfNy4TKyozC-_Ao9GP09ytNr

dl9rdz, 2021-03-24:
Interesting.This works fine on all versions I have seen so far...

Which software are you using? (binary devel image? something you compiled yourself?) Any strange configuration settings (TFT modeflip should be 0)? Do you have a link to the vendor of the display you are using?

What seems to happen is that the wrap-around for the high speed bitmap color transfer for the TFT9225 is somehow off-by-one. No idea how this can happen.

sebikolo, 2021-03-24:
I bought three LCDs. One works incorrectly. Two are working correctly. I am using devel20210315-B10-full.bin. 

https://pl.aliexpress.com/item/32952021835.html

dl9rdz, 2021-08-06:
Closing this issue for now. The google drive document does now longer exist, and possibly this is fixed with the new display driver anyway. If not, feel free to reopen.

closed 2021-08-06 by dl9rdz