dl9rdz/rdz_ttgo_sonde #70: Where is bluetooth connectivity?

chillmf, 2021-03-24:
This is a plea for help.

I have been able to get the code to compile and upload to an old TTGO T-Beam version 0.7 board.

The Access Point Internet Connection works well and there is traffic on the serial USB port.

I am not using a display and I have set the display option to 0 = old.

I have not been able to find a Bluetooth signal using my phone.

This is a 433 mhz board and I have been able to successfully compile and run the softRF application and to connect data via Bluetooth.

I have reviewed the code in the INO file and I find only one obscure reference to blue tooth.

I will appreciate receiving any advice on how to connect via Bluetooth.

Thanks,

Charley

dl9rdz, 2021-03-24:
I don't know what you want to achieve, but the short and simple answer is that there is no bluetooth connectivity...

chillmf, 2021-03-24:
Thanks ever so much for the quick and to the point reply.

Thanks for sharing all of your hard work!

Regards,

Charley

Links:
------
[1] 
https://github.com/dl9rdz/rdz_ttgo_sonde/issues/70#issuecomment-805874859
[2] 
https://github.com/

closed 2021-05-23 by dl9rdz