dl9rdz/rdz_ttgo_sonde #72: Meaning of WiFi mode

maehw, 2021-03-30:
This is not really a bug, rather a request for clarification

**Version and specific configuration:**
 - What firmware version are you using? master image http://rdzsonde.mooo.com/master/master_v0.8.0-B4-full.bin which seemed to be the newest release
 - Any specific settings: only added my local WiFi settings next to the AP settings
 - Hardware you are using? TTGO T-Beam ESP32 LoRa 433MHz REV1, no display, but should not be relevant

**Additional context**
What is the meaning of the options for the WiFi modes? In the web interface you can select 0/1/2/3 with a default of 3. I did not find any hint in the repo nor in the wiki (had expected it in https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Wifi-configuration). Is this setting irrelevant/obsolete?

dl9rdz, 2021-03-30:
See https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Software-configuration

dl9rdz, 2021-03-30:
Oh, I can add a minor detail that is not in the documentation. WiFi mode 1 runs in the background, and the ESP32 is a dual core CPU, so this will result in both cores doing dynamic memory allocation in parallel. This is basically fine, but may cause memory fragmentation, and in some rare cases may result in out-of-memory situations.  In my display driver, this may result in the display not being updated, and in the TCP/IP stack or the web server this may result in a crash.  This is very rare and usually should not be a problem. but if you see problems (most likely if you use the edit.html thingy to modify the screen layout) then you might want try wifi mode 3 instead of 1.

maehw, 2021-03-30:
Thanks for the clarification.

I did not see the forest for the trees. The documentation in https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Software-configuration is perfectly find, however I would have expected it in the https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Wifi-configuration article.

And I did not select the Wiki pages when searching the repo. https://github.com/dl9rdz/rdz_ttgo_sonde/search?q=wifi+mode&type=wikis

Sorry for the inconvenience, should have read the manual more carefully. ;)

closed 2021-04-02 by maehw