dl9rdz/rdz_ttgo_sonde #73: Map not showing in web UI

LukePrior, 2021-04-01:
When I got to SondeMap on the WebUI I see the following:

[image]

How is this supposed to work?

Thanks

Mooni123, 2021-04-02:
Sondemap Link/Site is no longer available and link is already removed in actual dev version of rdz_ttgo-sonde.

DL2MF, 2021-04-30:
The link to my old website wx.dl2mf.de  is outdated since more than 1 year. 
I'm currently working on a fixed version in my fork and will contribute a PR soon.

LukePrior, 2021-04-30:
> The link to my old website wx.dl2mf.de is outdated since more than 1 year.
> I'm currently working on a fixed version in my fork and will contribute a PR soon.

Exciting looking forward to it.

closed 2021-04-30 by LukePrior

LukePrior, 2021-06-18:
> I'm currently working on a fixed version in my fork and will contribute a PR soon.

Any progress on this?