dl9rdz/rdz_ttgo_sonde #74: Teile des Text spiegelverkehrt

do4bmw, 2021-04-06:
Hy

Ich habe da ein kleines Problem mit der Anzeige , 

verwendet habe ich das folgende bin Image devel20210402-B11-full.bin

Das Logo beim Booten wird normal angezeigt, doch der WiFi scan ist dann Spiegelverkehrt.

IP Addresse ebenfalls gespiegelt , die anderen Screens wieder wie die GPS Infos oder Akku z.b. korrekt angezeigt ...

Siehe bilder , was könnte das sein ?

https://imgur.com/gxLubj7
https://imgur.com/demV8pq
https://imgur.com/0nxslVl

dl9rdz, 2021-04-06:
Hi,

das ist jetzt der dritte solche Fall. Ein etwas mysteriöses Display. Verhält sich anders als die meisten ILI9225-Display. Ist das auch von ebay bestdeliverybuy?

Zur Ursachenforschung muss ich erst mal so ein Display in Hand bekommen. Scheint irgendwie so, dass der "schnelle" Bild-Transfer eines Bitmaps von dem Display falschherum verarbeitet wird.

do4bmw, 2021-04-06:
Hy,

ist dieses display

https://de.aliexpress.com/item/4000562221850.html?spm=a2g0s.9042311.0.0.6e824c4dI6R7uB

dl9rdz, 2021-04-06:
Ah ok danke. Und es sieht genauso wie auf dem Bild aus? Also das Flachbandkabel vom Display zur Platine auf der Seite von den 4 SD-Karten-Kontakten?

dl9rdz, 2021-04-06:
Just an additional comment in English:

I recently started getting reports about ILI9225 TFT displays with the output being somewhat wrong. It looks like fast bulk transfers of bitmaps end up in the wrong direction in the display's RAM. I have absolutely no clue about what is the cause for that, but I'll try to investigate.

The first display was bought somewhere on ebay and differs from my display by the display being mounted 180 rotated on the board (the ribbon cable from display to board being the side of the long pin header). The second display was bought on ebay from bestdeliveryboy (no more info available, the pictures on ebay show the ribbon cable on the side of the short pin header, but I don't know if that matches reality...). do4bwm's display might also look like its of the 180° rotated kind due to the slight misalignment of the display within the case. It is from aliexpress (see link above).

The displays I am using myself are from aliexpress, bought in 2019 from
https://de.aliexpress.com/item/33012793224.html?spm=a2g0s.9042311.0.0.56504c4dJlPXBG
(selecting color=2.0 inch)

**If you have such a problem (display output slightly in wrong positions / wrong direction / mirrored / whatever), please**

- **send me pictures of your display (ideally showing the complete board, front and back, and also the erroneous output on the display)**
- **let me know if you are using the binary image (and which version), and if you did change any display-related options in the config**
- **let me know where you did buy your display**

do4bmw, 2021-04-07:
> Ah ok danke. Und es sieht genauso wie auf dem Bild aus? Also das Flachbandkabel vom Display zur Platine auf der Seite von den 4 SD-Karten-Kontakten?

mmhhh ... das Flachbandkabel ist auf der Seite mit den vielen Pins und nicht auf der mit den 4 Pins ....

ansonsten stimmt die anzeige ... so wie es aussieht ist das problem nur bei der "kleinen" schriftart

https://imgur.com/jHzWwTw

do4bmw, 2021-04-07:
Here are some pictures of the PCB of my Display

Front = https://imgur.com/B6LEJkx
Back = https://imgur.com/LHwpZz4

Order Link = https://de.aliexpress.com/item/4000562221850.html?spm=a2g0s.9042311.0.0.6e824c4dI6R7uB

dl9rdz, 2021-04-07:
> ansonsten stimmt die anzeige ... so wie es aussieht ist das problem nur bei der "kleinen" schriftart

Es ist komplizierter. Das Display arbeitet irgendwie spiegelverkehrt. Die kleinen Schriftarten werden Punkt für Punkt ausgegeben, da sieht man das direkt. Bei den größeren Schriftarten definiert der Treiber einen rechteckigen Bereich, in dem die Textausgaben erfolgen wird. Dieser Bereich landet als ganzes gespiegelt an anderer Stelle. Das "Scan" sollte z.B. in der linken oberen Ecke stehen.  Der Inhalt des Rechtecks wird dann per Schnell-Transfer ans Display geschickt, und z.B. das "S" von Scan sollte ganz in der Ecke vom Display sein. Da wird das ganze irgendwie nochmal gespiegelt, so dass es lesbar aussieht. Aber der Text steht dann nicht ganz dort, wo er sein sollte.

do4bmw, 2021-04-17:
wenn du magst , kann ich dir so ein display zukommen lassen, könntest du behalten.

stimmt deine adresse bei qrz.com ? wenn ja werfe ich es am montag in die post

dl9rdz, 2021-04-18:
Habe inzwischen ein solches Display erhalten im Tausch gegen ein "gutes" erhalten.

Es ist seltsam. Verwendet man die Arduino_GFX-Library, funktionieren beide Displays 100% identisch. Bei der  Nkawu /
TFT_22_ILI9225 Library kommt es zu den beobachteten Problemen, nicht nur mit meiner Software, sondern auch mit den Standard-Demo-Anwendungen.

Die langfristige Lösung könnte also sein, die Library zu ersetzen. Hat nur noch zwei kleine Haken, zum einen ist der Display-Update langsamer (das kann man versuchen zu opimieren, erfordert aber ein wenig Zeit), und zum zweiten fehlt in der Arduino_GFX-Library eine Schriftart, die man nicht ganz einfach übernehmen kann.

do4bmw, 2021-04-18:
War nur ein angebot :-)

ich habe inzwischen auch ein "gutes" Display gekauft (Flachbandkabel auf der Seite der 4 Pins) und jetzt klappt es wunderbar :-)

dl9rdz, 2021-05-29:
Die aktuelle devel-Version sollte mit neuer Library mit allen Displays klar kommen, daher.... close

closed 2021-05-29 by dl9rdz