dl9rdz/rdz_ttgo_sonde #75: Scanner läuft weiter trotz -1

DL1LOL, 2021-04-07:
wenn im Scanmode eine Sonde gefunden wird bleibt er stehen wie er es soll
habe im Config Menue auf -1 gestellt 
werden aber keine Daten mehr von der Sonde empfangen
fängt er trotzdem wieder an zu scannen und bleibt bei anderen stehen
was zum Teil zur Verwirrung führt, bei der Sonden Jagd
nicht nur ich habe das Problem

DL2MF, 2021-05-23:
I've noticed the same issue on a TTGO T-Beam with latest devel here.

73 Mike, DL2MF

dl9rdz, 2021-05-23:
Very strange, I cannot reproduce this here. Already tried several different boards and software versions. Re-tried it right now, receiving S3130723 at the moment, disconnected the antenna, and the software does not switch back to scan mode.

The only known "issue" is that if you change the entry in the config to -1, **you have to do a reset of the TTGO**. Until you do the reset, the modified config will not be used.

If you have a log file from serial port (ideally including everything from reset to receiving a sonde to switching back to scan mode) send it directly to me dl9rdz@darc.de

DL2MF, 2021-05-23:
Tnx Hansi for your rapid reply and the reminder to make a reset.
Will try and confirm it tomorrow with next launches.

dl9rdz, 2021-08-06:
Not more input here, and I am unable to reproduce, so closing this issue.

Remeber to do a reset after stetting the config value to -1. The new value is only used after a reset.

closed 2021-08-06 by dl9rdz