dl9rdz/rdz_ttgo_sonde #78: Overlapping text

rmoravcik, 2021-05-24:
Hello,

today I upgraded to version devel20210524 and I observing text overlapping. Please, see the screenshot. I'm using TFT2" display with new alternative ALT9225 driver and screen3.txt.

[image]

dl9rdz, 2021-05-24:
Thanks for testing. 

I have updated the font to use less space, please try if this solves the problem already. 

You could also rearrange the lines in screens3.txt such that the display is drawn from top to bottom, then this overlap should not be a problem as well.

(I did not update the version number, update is in devel20210524)

rmoravcik, 2021-05-26:
Hello Hansi,

I just tested the latest version and it's working fine to me now. Thank you

closed 2021-05-26 by rmoravcik