dl9rdz/rdz_ttgo_sonde #88: Documentation

MikeTango, 2021-06-15:
Moin,

Is there any timeline to update the documentation? 

- several display options are not mentioned in the wiki
- the whole APRS upload thing is new and has server info pre-set, not mentioned in the wiki
- some wiki entries mention "below" and there is nohing below

I'd love to help with that, but i am struggling to make heads or tails of most of these things.

dl9rdz, 2021-06-15:
No strict time line.... Maybe in July/August it would be nice to have a master version 1.0, including up-to-date documentation.  

Help is certainly welcome. Small issues can be fixed fast, just let me know specifically what you are missing. Probably you mean MQTT, not APRS? The other new thing is Sondehub, but that is briefly mentioned in the wiki already. And still in testing and development, new features like the chase thing will first make it into the software, and then when working stable, to the wiki. Eventually.

DL2MF, 2021-08-17:
OK MikeTango,
let's start If you are not to busy to hunt radiosondes in all regions of DL. Please send any input to us if you like.