dl9rdz/rdz_ttgo_sonde #89: M10 callsigns uploaded incorrectly to Sondehub

darksidelemm, 2021-06-17:
It looks like in devel20210614 the M10 callsigns are being uploaded without the dashes between callsign sections, which results in a second balloon appearing on the tracker.

Example of uploaded data from devel20210614: http://api.v2.sondehub.org/sonde/011213333
Data uploaded by auto_rx stations: http://api.v2.sondehub.org/sonde/011-2-13333

dl9rdz, 2021-06-17:
Thanks for the info, should be fixed in devel20210618 (not tested, due to lack of time, but its just a small change, so chances are good I got it right I hope ;-))

Benoit37000, 2021-06-18:
Thanks dl9rdz

i have been playing with yours sketch since 4 month  :)

i made the mod for M10 and it is working today

see https://tracker.sondehub.org/?sondehub=1#!mt=roadmap&mz=8&qm=3_hours&f=011-2-13350

i jsut made those mod
see dif here
https://github.com/Benoit37000/rdz_ttgo_sonde/commit/8eaf666cdf79b99e4c1473ef600c7d4ee3a39c7e#diff-82075bbe53436f7aa132039429f5deb13f72e7203507e2b0764eda8973807555

String stringOne = s->ser;
"\"serial\": \"%c%c%c-%c%c-%c%c%c%c\","

stringOne[0],stringOne[1],stringOne[2],stringOne[3],stringOne[4],stringOne[5],stringOne[6],stringOne[7],stringOne[8]

Benoit F4EYW

dl9rdz, 2021-06-18:
With my change from yesterday evening, s->ser will already contain the serial number with the hyphens (changed in M10M20.cpp). The M20 serial is already generated with hyphens there, so it makes more sense to always use the full serial number with hyphens for M10 everywhere.

Benoit37000, 2021-06-18:
Oups  sorry
 i did not see you mod from yesterday , great  
:)

Benoit37000, 2021-06-18:
Thanks

Your Update Is great  M10  working on  Sondehub

dl9rdz, 2021-07-27:
Seems fine now, so closing the issue...

closed 2021-07-27 by dl9rdz