dl9rdz/rdz_ttgo_sonde #93: Last line in qrg.txt can´t be configured.

beliffm, 2021-06-22:
If the MAXSONDE parameter is set to n, only n-1 sondes can be configured. The last line always stays with 400.000 MHz regardless of any attempt to change it.

beliffm, 2021-07-30:
Just wondering. Did I step on someones foot?

dl9rdz, 2021-08-06:
No, thanks for the report.

"Its not a bug, it is just an unimplemented feature".  One entry was (long time ago) intended to be an entry to be changed dynamically (scanning, or some key presses to go up/down, or to copy peak from scanner, etc.). Has never been implemented, and is not intuitive, so I guess I should change it in the future.

beliffm, 2021-08-06:
Thanks for the update, i appreciate it.

dl9rdz, 2021-08-06:
In the latest version, this should be fixed now.

closed 2021-08-06 by dl9rdz